tjhorner/upsy-desky #3: Support for LINAK controllers

BradleyFord, 2022-08-29:
It would be great if it could support LINAK controllers, these are OEMed to many desk manufacturers and used an RJ45 connection. My IKEA desk is LINAK.

I will most likely buy a device from you when you launch, and can help provide whatever info you need to add support.

(Note their are a few github projects already for the LINAK)

tjhorner, 2022-08-29:
Can you link me to the projects for the LINAK desk? It will give me a good idea of whether support is feasible or not. Thanks!

BradleyFord, 2022-08-29:
The projects I had looked at are:
https://github.com/anetczuk/LinakDeskApp
https://github.com/rhyst/idasen-controller
https://github.com/j5lien/esphome-idasen-desk-controller

AABatteries, 2022-08-29:
+1 on this request, I was just about to ask about LINAK support :) I will be happy to help with hardware testing if need be.

jshep321, 2022-08-30:
Hi guys, I was thinking along the same lines at @tjhorner here!  

I measured my (~20yo) LINAK desk (from Kinnarps Sweden).  It was a 29.13V common that simply needs to be connected to wire A for up or wire B for down.  I bought a dirt cheap ESP01 dual relay from aliexpress and connected via ESPHome to HA.  

I made a capacitive touch sensor from a different ESP32 (that is part of my office master control) and can control the up/down with that via HA.

My desk has no height sensing so I am currently in the middle of prototyping a ToF distance sensor using HC-SR04 (ultrasonic) and a (3rd!) ESP.  I'm using node red to make Preset heights, but the sensor is currently too noisy to work properly.  I might switch over to a different position sensing method when I reopen this project.  I'm pretty surprised that everything is seamless except the ultrasound.  Delays are imperceptible in this setup.

Do your LINAK desks have position sensors?  My RJ11 is 3 wires only -- V+ (29v) and Up/Down sense.

Happy to share my yaml for the desk motor control if you want.

[image]

[image]

[image]

[image]

h42y, 2022-08-30:
There is high possibility that system is using hall quadrature encoders to keep the speed and position of the actuators in check. Just sniff them to get position.

tjhorner, 2022-09-06:
So it looks like all the existing projects use Bluetooth to communicate with the control box. I'd say that is the easiest path forward. Since the Upsy Desky has an ESP32 and thus Bluetooth, I recommend using [this ESPHome component](https://github.com/j5lien/esphome-idasen-desk-controller) for now while I look into what it uses for the physical connection.

AlexPotterOnGit, 2022-09-22:
+1 for LINAK. As others ready  to do tests required. I don't think my controller has Bluetooth so I am after wired connection.
[image]
[image]
[image]
[image]

Tam-Lin, 2022-10-07:
I have an uppy desky now and a linak controller. For this controller, there are 4 wires, and some communication between the desk and controller that happens to communicate the current height.

Tam-Lin, 2022-10-07:
I've done a bit of investigation.  There are 4 wires; from left to right:
1 2 3 x x x x 8

It looks like the wire color could be different, based on the picture above.  In my case, it's 
R B Y x x x x Bl

Measurements, which didn't  trigger the desk to do anything:

1 - 2:  -6.6 V
1 - 3: - 4.4 V
1 - 8: - 7.34 V

2 - 3: 0 V
2 - 8: 0 V

3 - 8: 0 V 

If I do attach the upsy desky to the desk, it descends to the bottom.

I have a Linak DPG1C controller; I'm not sure about the motor control box.

pi2madhatter, 2023-02-22:
I would throw down the money here and now for LINAK support. I have one that looks identical to the others posted on this, but with slightly different specs. I've attached an image of my label if it helps at all.

[image]

Ordspilleren, 2023-08-28:
Hi all!
I have been working on exactly this, and I am happy to report that I have a working solution which you can find [in this repo](https://github.com/Ordspilleren/esp32-linak-desk-control). I didn't do this all by myself, so I have given credit where its due.

Integrating it directly within Upsy Desky is unlikely in its current form, simply because I believe esp-idf is a requirement to get the control we need over UART.

It's by no means done yet, but it works reliably for me.

Ordspilleren, 2023-08-28:
> @Ordspilleren You can actually compile your firmware directly for the Upsy Desky if you want; it's just an ESP32 module.

Good point, that's certainly an option if someone wants to go that route. They would still need the custom circuit described in my repo however, unless Upsy Desky can support doing single wire RX and TX and bring the 12V down to 3.3V. I haven't looked at the options in-depth to be honest.

> Why do you think ESP-IDF is required? ESPHome [does support ESP-IDF](https://esphome.io/components/esp32.html#esp32-espidf-framework) but currently it's using the Arduino framework, which is default.

My solution relies on detecting a UART break signal and then clearing the UART FIFO buffer to get the first data byte in the data frame from the table. I also set a couple of ESP-IDF specific options to get the break detection working reliably. Not sure if that is possible with Arduino directly.

With that said, you _can_ call IDF functions from within Arduino, so I guess it should be possible if someone was willing to put in the time.

davidski, 2023-09-28:
It looks like HA is getting 1st party integration with IKEA IDASEN in the 2023.10, which should (not yet tested) include Linak devices. I'm excited to see this land! https://rc.home-assistant.io/integrations/idasen_desk

Ordspilleren, 2023-09-28:
> It looks like HA is getting 1st party integration with IKEA IDASEN in the 2023.10, which should (not yet tested) include Linak devices. I'm excited to see this land! https://rc.home-assistant.io/integrations/idasen_desk

Please note that this integration works via the Bluetooth functionality present in some Linak controllers. This has been possible for several years via [esphome-idasen-desk-controller](https://github.com/j5lien/esphome-idasen-desk-controller) and [LinakDeskEsp32Controller](https://github.com/krzmaz/LinakDeskEsp32Controller).

giantorth, 2024-02-27:
Has anybody gotten Linak support working with an upsy yet?  I suppose I should have checked the issues before I bought one of these ;)

giantorth, 2024-02-27:
I'm not sure this is helpful, but the Linak desks did have a USB adapter for PC control, the USB2LIN06 cable.
You can see pictures of it here: https://github.com/UrbanskiDawid/usb2lin06-HID-in-linux-for-LINAK-Desk-Control-Cable/tree/master/PHOTOS/usb2lin06-HID

I'm curious if it would be possible to use this cable as some sort of bridge for the hardware compatibility issues with the upsy and a linak controller, perhaps with a bit of soldering if necessary.

giantorth, 2025-03-05:
Not sure who's still got these Linak desks, let alone the USB adapter for one, but for those wanting to smarten up these ancient Linak controllers I've written an MQTT service to control my desk using a pi zero and a USB2LIN06 adapter: https://github.com/giantorth/linak-mqtt-ctrl