tjhorner/upsy-desky #5: Adapter Cable Research

tjhorner, 2022-08-31:
There are many different standing desk brands out there, and most of them use the RJ45 pinout that this project does. However, there are a few that do not! How do we add support for them? I think the best solution is to create adapter cables which can easily and cheaply modify the pinout or physical connector so that the Upsy Desky can communicate with the desk.

For example

- LoctekMotion desks
  - Uses RJ45
  - Different pinout (see [here](https://github.com/tjhorner/wifi-desk-controller/discussions/2#discussion-4151398))
  - Solved with a custom cable which is wired differently at one of the terminating ends
- Autonomous.ai desks
  - Uses RJ50
  - Different pinout
  - Solved with a custom cable which terminates in RJ45 at one end and RJ50 at the other

I think this is a good way to solve the problem, but I'm not entirely sure how to manufacture these at scale. I'll be researching different ways to do this.

tjhorner, 2022-09-02:
## Autonomous Desk Adapter
- **Connector**: RJ50 (10p10c)
- **Pins in use**: 5

### Pin Functions

| Position | Color  | Function |
|----------|--------|----------|
| 1        | N/A    | N/A      |
| 2        | N/A    | N/A      |
| 3        | N/A    | N/A      |
| 4        | Red    | Sleep    |
| 5        | N/A    | N/A      |
| 6        | N/A    | N/A      |
| 7        | Orange | GND      |
| 8        | Yellow | RX       |
| 9        | Green  | TX       |
| 10       | Blue   | +5V       |

### Pin Mapping

| Function | RJ45 Position | RJ50 Position |
|----------|---------------|---------------|
| Sleep    | 1             | 4             |
| RX       | 2             | 8             |
| GND      | 3             | 7             |
| TX       | 4             | 9             |
| +5V      | 5             | 10            |
| N/A      | 6             | N/A           |
| N/A      | 7             | N/A           |
| N/A      | 8             | N/A           |

tjhorner, 2022-09-02:
Reached out to a cable manufacturer to get prototypes of the Autonomous adapter cables made based on the above info.

rosstadude, 2022-09-13:
I would be interested and willing to test on my Autonomous desk. Please let me know how much for the cable. Thank you

Jolanrensen, 2022-10-01:
Also willing to test on my locktech desk! I don't have an upsy desky yet, but I'm on the waiting list

Jolanrensen, 2022-11-03:
I now do have one :) Also, I have some ethernet (RJ45) plugs and a crimping tool, so I can modify and test a cable to see if it works on HS01B-1 Locktech desks. Did follow [this](https://github.com/tjhorner/wifi-desk-controller/discussions/2#discussion-4151398) but it seems to have gone silent.
I'm also curious if one or two cables would need to be modified. Since the upsy desky is a passthrough I think the cable to the controller/screen of the desk would also need to be modified as well as a cable to the desk motor itself. Is that right?

tjhorner, 2022-11-03:
Yes, getting the original keypad to work will be tricky and might require some kind of other adapter. For now I am focusing on getting the Upsy Desky working with the control box

Jolanrensen, 2022-11-04:
Alright :) If ever needed, I did find some extra info about passthrough for the keypad: https://github.com/iMicknl/LoctekMotion_IoT/tree/main/packages/esphome#hs01b-1-with-esp32-as-passthrough-and-keypad-support

Jolanrensen, 2022-11-05:
Alright, so I created an adapter cable based on the sources available to me:

RJ45 OG | cat5 OG colors (upsy desky side) | RJ45 Target | Cat5 Target colors (desk side) | HS01B-1 purpose (after rewire)
-- | -- | -- | -- | --
1 | orange/white | 1 | orange/white | RES (needs passthrough for keypad)
2 | orange | 6 | green | RX (needs bridging for keypad)
3 | green/white | 7 | brown/white | GND (needs bridging for keypad)
4 | blue | 8 | brown | TX (keypad should have different GPIO)
5 | blue/white | 2 | orange | 5V (needs bridging for keypad)
6 | green | 4 | blue | SWIM (needs passthrough for keypad)
7 | brown/white | 3 | green/white | (unknown) (needs passthrough for keypad)
8 | brown | 5 | blue/white | PIN 20 (keypad should have different GPIO)

Some clarification: "Passthrough" can mean either ouside the upsy-desky or through it without touching the signal. I don't know if it's needed for the desk to operate but it's what mentioned here https://github.com/iMicknl/LoctekMotion_IoT/tree/main/packages/esphome#hs01b-1
"Bridging" means the upsy desky needs the connection but it also needs to be passed through to the keypad.
Not sure what needs to be done with TX and PIN 20 since they are wired to different GPIOs altogether, both from/to the keypad as from/to the desk controller.

But let's start with just the desk controller as I created the cable for it. Connecting the upsy desky gives it power (since it now has 5v at pin 5), but where do I go from here? I have programming experience for sure, but haven't worked with esphome before, so if I could be pointed in the right direction, that would be great :)

tjhorner, 2022-11-05:
@Jolanrensen excellent research! To test out the firmware you can follow ESPHome's [setup guide](https://esphome.io/guides/getting_started_command_line.html) then try installing the various configs from the repo you linked above ([here](https://github.com/iMicknl/LoctekMotion_IoT/tree/main/packages/esphome)).

Jolanrensen, 2022-11-06:
So the plan is to get that to work as best as possible and then adapt your template to it? :) Cause that sounds doable. I'll report back when I have something working!

Jolanrensen, 2022-11-07:
Small update: I'm able to control the desk from home assistant! (using https://github.com/Jolanrensen/LoctekMotion_IoT) It does seem to stutter a lot XD Not yet sure if that's because of my wiring or the software.

Jolanrensen, 2022-11-08:
Just as a redundancy test, connecting a similar adapter from they keypad to the upsy desky does light up the keypad and displays the correct height, but the desk is uncontrollable, both from HASS, as well as the buttons. Thought so :) 
However, to be able to connect both the desk and the keypad to the upsy desky I probably have to make another adapter cable, disconnecting a couple of lanes to free up GPIO ports

Jolanrensen, 2022-11-15:
@ajfriesen That's exactly what I ordered :) 
But I would still need a sort of special cable-adapter since some cables need to bypass the upsy desky going straight from the keypad to the desk and some need to go through.

Something like (forgive my drawing):
[image]
Where 
UD: upsy desky
K: Keypad
b: Desk

But hopefully I can soon do some more testing :)

tjhorner, 2022-11-16:
> Just an idea: For mass production it may be easier to look into an adapter board where you punch down cables by yourself?

That's a great idea! Never thought of that. I'll see how the cost compares to custom cables.

Jolanrensen, 2022-11-17:
> @ajfriesen That's exactly what I ordered :) But I would still need a sort of special cable-adapter since some cables need to bypass the upsy desky going straight from the keypad to the desk and some [...]

Well, I've got something working :D Using my branch of LocTekMotion https://github.com/Jolanrensen/LoctekMotion_IoT/tree/main I can now control the desk both from the keypad and home assistant :)
There are still some problems: 
  - The screen just stays on, not sure why. 
  - When the UD just booted, the desk height is NaN until changed, but sometimes the UART gets flooded with 0x00 freezing up and crashing the UD
   - The number buttons and M/A do nothing with the desk. They do get picked up by the software and, in theory, they send their command to the desk, but nothing happens.

However, these problems are most probably software problems :)
My wiring (hopefully including all coloring standards, haha):

cat5 B | cat5 B | HS01B-1 | Upsy Desky side (from desk) | Upsy Desky side (from keypad)
-- | -- | -- | -- | --
1 | orange-white | brown |   |  
2 | orange | white | 5/blue-white/green (RX GPIO17) | 5/blue-white/green (RX bridged from desk GPIO17)
3 | green-white | purple | 7/brown-white/blue (GND) | 7/brown-white/blue (GND)
4 | blue | red | 6/green/black (TX GPIO16) |  
5 | blue-white | green | 8/brown/yellow (5V (VVD)) | 8/brown/yellow (5V (VVD))
6 | green | black |   | 6/green/black (TX or RX from UD pov GPIO19)
7 | brown-white | blue |   | 4/blue/red (pin 20 GPIO21)
8 | brown | yellow | 4/blue/red (pin 20 GPIO22) |  

Connected straight from desk to keypad:
  - 1 / orange-white / brown
  - 2 / orange / white
  - 3 / green-white / purple

Yes, I changed my wiring method a bit to make it easier to work with for my breadboard.

Jolanrensen, 2022-11-19:
My beautiful adapter! It's completely based on the table above.
[image]

Simply plug the two male plugs in the Upsy Desky and connect the female plug to the keypad and the male plug in the desk controller:
[image]

It's easy to replicate if you have a small ethernet extension cable (male to female, adhering to cat5 B) where you strip out the middle bit, cut all wires except 1,2, and 3 and then attach two male ethernet plugs using the table above.
If you don't have an ethernet crimping tool, there are also toolless options available:
[image]

Not sure if it would be possible to manufacture something like this, but doing it yourself can be done in less than an hour :)

rosstadude, 2022-11-19:
This is great!

Jolanrensen, 2023-01-19:
@MrTechGadget 108ms is very specific timing indeed haha. Very cool! I will definitely try that too. I'm currently using a delay of 0ms which is smooth but sometimes "hesitates" a bit. I do think that these timings are very specific to your own setup on the cables, connectors, the motor etc.

backslashV, 2024-01-02:
Hi there, I have a similar desk with an RJ50 keypad connector. I'd like to use an Upsy Desky to make the desk smart, but that device uses an RJ45 connector. Can anyone suggest a way to identify what each of these 5 pins do so I can make a custom RJ50 to RJ45 adapter?
[image]

tim-vk, 2024-07-24:
Silly question: but what is the pinout that the upsy desky expects? 
In the documentation it mentions pin 3 = GND and pin 5 = 5v. But which ones are the RX & TX pins? I couldn't find that in the docs (and i'll probably need to open the pcb files: which i can't).

JesperKuipers, 2024-09-29:
> Silly question: but what is the pinout that the upsy desky expects? In the documentation it mentions pin 3 = GND and pin 5 = 5v. But which ones are the RX & TX pins? I couldn't find that in the docs (and i'll probably need to open the pcb files: which i can't).

https://github.com/tjhorner/upsy-desky/blob/master/firmware/base.yaml
line 19 and line 35