tjhorner/upsy-desky #6: Unable to operate Autonomous desk

ChrisCrewdson, 2022-09-04:
Perhaps it uses a different protocol?
I cut the controller cable with the RJ50 and spliced in an RJ45 on both cut ends, following the pinouts from https://github.com/tjhorner/upsy-desky/issues/5#issuecomment-1235040100 
It seems like it should be fine since power and ground at least seem right with multimeter testing.
It does still work with the controller when using two keystone jacks punched down back to back.

When I attach the upsy-desky to the desk, it powers on and I can see it in home assistant, but it does not control the desk or read height.

When plugging in the controller to the upsy-desky, the controller display shows full digits and the backlight powers on
However, the buttons only make the backlight turn on while pressed (the normal behavior), but do not operate the desk and the height is not shown. 

The controller is https://www.timotion.com/en/products/controls/tdh6-series and the box is https://www.timotion.com/en/products/control-boxes/tc15-series

tjhorner, 2022-09-04:
The protocol is indeed different. Someone else has reverse engineered the protocol the Autonomous desks use and posted their work in this repo: https://github.com/Stefichen5/AutonomousControl

I do plan on adding official support for Autonomous desks very soon! Need to make the adapter cables then work on adding support in the firmware.

ChrisCrewdson, 2022-09-04:
Thank you! I had just found this discussion which lead me to the same place: https://github.com/tjhorner/wifi-desk-controller/discussions/16

Gast0n87, 2022-09-22:
> Thank you! I had just found this discussion which lead me to the same place: [tjhorner/wifi-desk-controller#16](https://github.com/tjhorner/wifi-desk-controller/discussions/16)

Did you manage to get it working?

ChrisCrewdson, 2022-09-22:
I have not managed to get this working. My first step was to attempt to log the periodic messages from the desk to the controller, but nothing has come through. There might even be a hardware problem since I think the two RJ45 ports on the upsy-desky are tied together and they may need to be separate UARTs: https://github.com/tjhorner/wifi-desk-controller/discussions/16#discussioncomment-3090583
That being said, I haven't managed to get anything working even with just one of the desk or the controller attached at a time.

normanr, 2022-12-31:
fwiw EasyErgo also uses the TiMotion devices.
I have a TC15S control box with a TDH18P controller. Pins are 2,4,7,8,9,10 (black,red,orange,yellow,green,blue) which matches the opening comment (extra black wire on pin 2 though). I haven't been able to verify signals yet, but I'm hoping that a USB to serial cable can be made to work.

normanr, 2022-12-31:
I noticed that the TDH18P has a RJ50 jack in the back of the unit (for a bluetooth dongle, model TWD1), so I checked with a multi-meter and confirmed that pins of the jack seem to be directly connected to the plug. Referring to the functions documented at https://github.com/tjhorner/upsy-desky/issues/5#issuecomment-1235040100 I also confirmed that GND (Orange) and +5V (Blue) are directly connected to the USB plug GND and VCC pins.

I checked the pin voltages at the jack on the TDH18P while it's plugged into the control box and (unsurprisingly) they match the documented functions. RX and TX are near +4.8V and change slightly when the display updates (RX), or when buttons are pushed (TX), they seem to return to +5V when the display is off. Sleep (Red) is +5V when the display is off, and 0.2V when the display is active. Pin 2 (Black) is +5V, but I have no idea what it's used for.

If I plug an cat5 cable directly into the TC15S control box (i.e. with the TDH18P controller disconnected), then pins 3, 5, and 6 also have 5V (and TX and RX are +5V, so no data is being sent or received when the control unit is not attached). Based on the PDFs linked to from https://www.reddit.com/r/esp8266/comments/vlu0om/beginner_standing_desk_project/ I tried shorting pin 7 (GND) to pins 5 and 6 and confirmed that they cause the desk to move up and down. I didn't try the same with pin 3 (and didn't have access to pin 1, because the cat5 cable only connects to pin 2 through pin 9).

normanr, 2022-12-31:
I found some internal photos of the bluetooth dongle in the [fcc filing](https://fcc.report/FCC-ID/W6JTWD1) which show that only pins 4, 8, 9, and 10 are used, so that's nice confirmation too.

Andrerm124, 2023-01-02:
Very interested in this topic, I'm not experienced in this field so can't really contribute, but @normanr if you find anything it'd be really interesting if you could share 🙂

JohnAZoidberg, 2023-01-09:
I've got a table with TDH20P handset and TC15S Control Box.
The UART pins match https://github.com/tjhorner/upsy-desky/issues/5#issuecomment-1235040100
And the protocol matches https://github.com/adfinis/muuvctl#serial-protocol

JohnAZoidberg, 2023-01-10:
To clarify what I did:
I got RJ50 (10P10C) and RJ45 (8P8C) connectors and crimped them together on a cable.
Then using an RJ45 breakout board I connected the TX,RX and GND pins to a cheap logic analyzer.
I plugged RJ50 into the back of the handset.

After confirming the protocol looks fine, I connected it to a UART to USB adapter.
This would cause issues when I connected something to TX on the adapter, so I first connected the cables's TX to the adapter's RX and then the cable's RX. With pyserial i could snoop the traffic.

muuvctl doesn't seem to work solidly when connecting RJ50 to the control box. It can get the height sometimes but usually it displays the wrong byte.

testventure, 2023-02-14:
Dang should have checked here before I bought it.  But I will eagerly await an update to handle the TiMotion desk.

StillAlyve, 2023-04-22:
I've also got a TC15s controll box with pins, that match [these](https://github.com/tjhorner/upsy-desky/issues/5#issuecomment-1235040100)  on the handset. 

Would I be able to run the python code on a pi using a console cable?

ericfri, 2023-08-11:
Any updates on adding Autonomous desk support?

tjhorner, 2023-08-11:
Unfortunately I haven't had the bandwidth to investigate this further as I've been very busy at my dayjob, and I don't have any Autonomous desk hardware to properly test it. However, Autonomous desk support is still something I would like to add in the future. I appreciate the efforts of everyone who has contributed in this thread — all the info presented here looks very useful for when I am able to pick this back up.