tjhorner/upsy-desky #7: Geekdesk Compatibility

J-Camping, 2022-09-05:
Hello! I recently received the Upsy desky and would like to try it out on my geekdesk.
It looks like pin 3 has 5v and pins 1,5,7,8 all appear to be grounds. 

The controller is labelled ey-h6032c, and does also have an rj11 port. I found this manufacturer page, but it didn't seem to have much information. https://www.eyoungonline.com/products/pro_51.html 

Guessing at a minimum I would need to make a custom cat 5 cable to swap the 5v and gnd to match the upsy desky?

Want check with you before breaking anything too badly.

Thanks,

James

tjhorner, 2022-09-05:
Hi! Yes, at minimum you would need to make the +5V and GND match the pinout on the Upsy Desky. The pinout is available [here](https://github.com/tjhorner/upsy-desky/wiki/GPIO-Pinout).

After that I would go ahead and try out the stock firmware and see if any of the existing protocol decoders work. If not, then more reverse engineering work is probably required.

J-Camping, 2022-09-07:
I reversed the wires for 5v and gnd, the board powers up and I can connect to the web interface. As soon as it turns on, the board goes to its highest or lowest level. I haven't moved any other wires yet.

When either uplift or jarvis is selected and I select preset 1 it goes to the bottom, preset 2 doesn't do anything, preset 3 goes to bottom, preset goes to 4,  and the slider doesn't do anything. 

Let me know what I can do to help with any reverse engineering, and thank you for creating this and the help!

Handset photos attached. 
[image]
[image]
[image]

cturra, 2023-02-16:
is there anything i can do to help out? happy to pull things apart on mine also.  i’m really interested in having my geekdesk supported

thomascantrellsea, 2023-10-27:
I wanted to record a few things I worked out:
- Do not plug your Upsy Desky into the Geekdesk MAX (EY-H6032C) without an adapter cable. I fried the Upsy Desky pretty quickly doing this. Desk is doing fine. It's my bad, I'll order another. This is very likely because of the differences in the 5V/GND pins.
- If pin 3 and pin 5 are flipped, it seems likely that one can use Upsy Desky though looking at the pinout. 

Pinout from my EY-H6032C control panel cable matched up with your image (JCH35-TLD-V3)

1. White - PTC1
2. Yellow - PTC2
3. Red - VCC = +5V
4. Green - PTB5
5. Black - GND
6. Blue - PTB6
7. Manual up - not connected on cable - see [1]
8. Manual down - not connected on cable - see [1]

Notes:

1. Jumping 3 to 7 or 8 causes the desk to go up and down. This matches with https://github.com/Hypnopompia/photon-desk/blob/master/README.md 
2. Also see https://forums.adafruit.com/viewtopic.php?p=649290, though the underlining control panel processor seems slightly different (JCH35-TLD-V2)
3. Appears that Jiecang has slightly different wirings between the Geek Desk and other desks.