tjhorner/upsy-desky #8: Support for wn17cm2 controller

AnalogAyub, 2022-09-05:
I was able to set up the UpsyDesky easily, but the unit would not power on with just the RJ45 from the desk controller box. Once connecting a USB-C, the unit was detected by Home Assistant but would not control the desk or detect a height. Additionally, the desk keypad daisy-chained through the UpsyDesky would not control the desk.

Here's what the box looks like: 
[image]

tjhorner, 2022-09-06:
Interesting, could you verify that the +5V and GND lines are on the correct positions? You can find the RJ45 pinout [here](https://github.com/tjhorner/upsy-desky/wiki/GPIO-Pinout). If they aren't in the correct place, you may need a special cable.

seanford, 2022-09-06:
Same for the WN17CM3, which I assume will have the same pinout.  This is the controller for the Monoprice desks.

seanford, 2022-09-06:
https://www.kontive.com/products/lifting_table_part/17.html

Maybe they can help.

AnalogAyub, 2022-09-07:
> Interesting, could you verify that the +5V and GND lines are on the correct positions? You can find the RJ45 pinout [here](https://github.com/tjhorner/upsy-desky/wiki/GPIO-Pinout). If they aren't in the correct place, you may need a special cable.

Honestly, I'm out of my depth here 😬

Providing power through USB-C was no problem, though.

seanford, 2022-09-26:
If anyone figures out the pin out for the controller, I can make a cable and test it out.

seanford, 2022-10-15:
Any instructions on how to test the pinouts so I can help move this along?

tjhorner, 2022-10-16:
@seanford I recommend getting an RJ45 breakout board [like this one](https://www.amazon.com/Treedix-Connector-Breakout-Compatible-Unassambled/dp/B08HGS5G8N) so you have somewhere to attach multimeter leads. Then follow this guide: https://github.com/tjhorner/wifi-desk-controller/wiki/Desk-Compatibility

Basically, just use a multimeter to find the position of the +5V and GND pins. After that we can start figuring out which pins transfer which data, if it uses serial, etc.

Jaharmi, 2022-11-18:
I have a desk with a WN17CM3 controller. The control pad is wired with an RJ45. I have found the following pinouts and color coding.

**Control panel wires**

|  **Wire color** | **Pin label** |
|  ----- | --- |
| Red | Vin |
| Blue | Rx |
| Yellow | Tx |
| Black | Gnd |

**RJ45 pinout**

| **RJ45 pin** | **Wire color** | **Note** |
|  ----- | ----- | ----- |
| 1 | Orange | |
| 2 | Blue | Rx from control panel |
| 3 | Black | Gnd from control panel |
| 4 | Yellow | Tx from control panel |
| 5 | Red | Vin from control panel |
| 6 | Orange | |
| 7 | Orange | |
| 8 | Orange | |

Below are pictures of my evidence:

1. the disassembled control pad, where the circuit board labels the wires
[image]

2. the RJ45 cable coming from the control pad (bottom and top, since it was difficult to see the colors on all the wires from the bottom)
[image]
[image]

Jaharmi, 2023-01-31:
With my just-arrived Upsy Desky, I see the same thing as @AnalogAyub.

1. I was able to set up the UpsyDesky easily.
2. Connected USB-C, the unit was adopted by ESPHome and detected by Home Assistant.
3. The unit would not control the desk or detect a height. I tried both decoders.
4. The desk keypad daisy-chained through the Upsy Desky would not control the desk.
5. The unit stopped responding with just the RJ45 from the desk controller box. I tried connecting to the Upsy Desky Web interface and it timed out.

I will double-check for the +5V and GND pins from the controller box.

Once I've found that, will the device need a debug firmware installed? Or is there something I can change in the (very minimal) ESPHome config?

Jaharmi, 2023-01-31:
I think RJ45 pin 2 is +5v and 3 is GND. I've also posted in the reverse engineering discussions area.

swoga, 2023-08-28:
i reverse enginieered the pinout and protocol some time ago, maybe this will help you: https://github.com/swoga/standingdesk

Jaharmi, 2024-02-02:
@swoga Awesome. I’ve been wondering about this for years! Do you have any thoughts on how this could integrate into Upsy-Desky / ESPHome?

swoga, 2024-02-02:
@Jaharmi
My plan is to write a custom component for ESPHome, but I haven't gotten around to it yet. It's definitely still on my to-do list.

Jaharmi, 2024-02-02:
@swoga If this fit in with Upsy-Desky, I’d be a fan. :) That would give it things like the Upsy-Desky local Web interface. I already have an Upsy-Desky and just never figured out how to get from beyond the pinouts to do anything else.

Looking at the [related component](https://github.com/tjhorner/esphome-standing-desk), the C++ part is currently beyond me.

Jaharmi, 2024-02-03:
@swoga Does this state diagram seem to handle what you found in the serial protocol for the WN17CM3? I am trying to understand the protocol and maybe do something with it. This is probably my first state diagram ever so I may have mixed it with a flow chart.

```mermaid
stateDiagram
    [*] --> CMDBEGIN
    CMDBEGIN --> CMD
    CMD --> DISPLAY: D
    CMD --> KEY: K
    CMD --> REQUEST: R
    CMD --> ACKNOWLEDGE: A
    DISPLAY --> HEIGHT
    HEIGHT --> HEIGHTCKSUM
    HEIGHTCKSUM --> CMDEND
    HEIGHT --> COMMAOPT
    COMMAOPT --> HEIGHTCKSUM
    KEY --> DONTCARE
    REQUEST --> DONTCARE
    ACKNOWLEDGE --> DONTCARE
    DONTCARE --> CMDEND
    CMDEND --> CHKSUM
```

Jaharmi, 2024-02-03:
@swoga Does this state diagram seem to handle what you found in the serial protocol for the WN17CM3? I am trying to understand the protocol and maybe do something with it. This is probably my first state diagram ever so I may have mixed it with a flow chart.

```mermaid
stateDiagram
    [*] --> CMDBEGIN
    CMDBEGIN --> CMD: colon
    CMD --> DISPLAY: D
    CMD --> KEY: K
    CMD --> REQUEST: R
    CMD --> ACKNOWLEDGE: A
    DISPLAY --> HEIGHT
    HEIGHT --> HEIGHTCKSUM: decimal number as string
    HEIGHTCKSUM --> CMDEND: semicolon
    HEIGHT --> COMMAOPT: comma
    COMMAOPT --> HEIGHTCKSUM
    KEY --> DONTCARE
    REQUEST --> DONTCARE
    ACKNOWLEDGE --> DONTCARE
    DONTCARE --> CMDEND: semicolon
    CMDEND --> CHKSUM
```

seanford, 2024-02-05:
> Vin from control panel

@tjhorner it seems that the pinouts for this controller are the same as other controllers, but the protocol is different.  With the information provided by @swoga, is it possible to write a new decoder for this controller?   I'd do it myself if I thought I could...lol

KiloHotl, 2024-02-06:
Hi, I recently purchased the WN20CM2NR V1.33 (printed on the PCB). The RJ45 on the control panel only has 4 wires:

<p dir="auto" style="box-sizing: border-box; margin-top: 0px; margin-bottom: 16px;"><strong style="box-sizing: border-box; font-weight: var(--base-text-weight-semibold, 600);">RJ45 pinout</strong></p>

RJ45 pin | Wire color | Note
-- | -- | --
1 |  |  
2 | Black | Rx from control panel
3 | Green | Gnd from control panel
4 | White | Tx from control panel
5 | Red     | Vin from control panel
6 |  |  
7 |  |  
8 |  |

Using a D1 mini and some research, I was able to intercept the UART comms but using the @swoga baud rate (19200), the data is different. This is the log:

```
[uart_debug:114]: <<< 3A:57:4E:31:38:48:33:32:2E:56:32:2E:30:30:2E:31:39:31:38:2E:30:30:31:42:46:3B
[uart_debug:114]: >>> 3A:4C:4C:39:38:3B
[uart_debug:114]: <<< 3A:57:4E:31:38:48:33:32:2E:56:32:2E:30:30:2E:31:39:31:38:2E:30:30:31:42:46:3B:3A:41:34:31:3B
[uart_debug:114]: <<< 3A:41:34:31:3B
[uart_debug:114]: >>> 3A:44:31:30:39:44:45:3B
[uart_debug:114]: >>> 3A:41:34:31:3B:3A:44:31:31:35:44:42:3B
[uart_debug:114]: <<< 3A:4B:20:33:4D:45:42:3B:3A:41:34:31:3B
[uart_debug:114]: >>> 3A:41:34:31:3B
[uart_debug:114]: <<< 3A:4B:20:33:42:45:30:3B
[uart_debug:114]: >>> 3A:44:31:31:35:44:42:3B
[uart_debug:114]: <<< 3A:41:34:31:3B
[uart_debug:114]: >>> 3A:41:34:31:3B:3A:44:31:30:39:44:45:3B
[uart_debug:114]: <<< 3A:4B:20:32:4D:45:41:3B:3A:41:34:31:3B
[uart_debug:114]: >>> 3A:41:34:31:3B
[uart_debug:114]: <<< 3A:4B:20:32:42:44:46:3B
[uart_debug:114]: >>> 3A:44:31:30:39:44:45:3B
[uart_debug:114]: <<< 3A:41:34:31:3B
```
I´m tryong to analyze the data in order to identify what`s going on

KiloHotl, 2024-04-05:
hi @swoga could yo develop the custom component? I been trying for several weeks and nothing worked.

iambucketdotcom, 2024-08-23:
Hi! I, too have this controller and have only been able ot get the UART messages to work with upsy desky. am I understanding correctly that more is still needed to make this unit work?

Thanks - Bucket

seanford, 2024-08-23:
Unfortunately, I don't think it is going to happen.

seanford, 2024-11-29:
Is it safe to say that this control box will no be supported by Upsy Desky?  Just wanted to know if I should continue to monitor this project for an update or just move on.

Jaharmi, 2024-12-15:
Unless someone like @swoga can figure out a component that handles the communication, I don’t see it happening. Although I’m very interested in making it work, it’s beyond my current understanding.

I’m still not sure why the Upsy-Desky doesn’t seem to get power from the WN17CM3 controller. It seems like _that_, at least, should be working, given the pinouts we’ve all found.

I am not sure whether or not my Upsy-Desky’s RJ45 ports even work.

I also don’t know if there’s a way to capture and decode the serial/UART communication. Maybe something could be done with the UART Bus component in ESPHome. 🤷

Jaharmi, 2025-01-04:
**Update:** I set up the reverse engineering firmware with the UART settings from @swoga. Since I could not set a `stop_bit` of `0`, I set it to `1`.

With this firmware, I was able to capture sequences that appear to correspond to the investigation by @swoga. I can confirm that the display on the connected keypad — which initially started out at `8.8.8` — did update with the expected desk height values that I normally see when the Upsy-Desky is not in the loop.

I am attaching the following files:
- firmware YAML I used: [upsy-desky-WN17CM3-reverse-engineering-uart-debug.yaml.txt](https://github.com/user-attachments/files/18307881/upsy-desky-WN17CM3-reverse-engineering-uart-debug.yaml.txt)
- log data showing the commands going back and forth successfully: [upsy-desky-WN17CM3-uart-logs-all-buttons.txt](https://github.com/user-attachments/files/18307880/upsy-desky-WN17CM3-uart-logs-all-buttons.txt)

Rough sequence of events from that test:

1. Flashed firmware to Upsy-Desky via USB and let it start up until I could see logs
2. Connected WN17CM3 controller (standard Ethernet cable with 2x RJ45) to Upsy-Desky
3. Connected keypad with its cable (RJ45) to Upsy-Desky
4. Keypad displayed "8.8.8" (I don't think the display changed during the testing)
5. Pressed Up button
6. Pressed Down button
7. Pressed preset button "1" and pressed it again to stop
8. Pressed preset button "2" and pressed it again to stop
9. Pressed preset button "3" and pressed it again to stop
10. Pressed preset button "4" and pressed it again to stop
11. Pressed button "M"
12. Disconnected controller and keypad from Upsy Desky

Jaharmi, 2025-01-04:
I refreshed the Upsy-Desky with similar firmware to what I used above, but changed back to the reverse engineering firmware’s `log_hex` lambda, rather than `log_string`. I didn’t make any other changes.

Here is the log capturing in hex: [upsy-desky-WN17CM3-uart-hex-logs-all-buttons.txt](https://github.com/user-attachments/files/18307921/upsy-desky-WN17CM3-uart-hex-logs-all-buttons.txt)

I followed the same rough sequence, trying to press each button at least twice. The display correctly showed the desk height when I connected the keypad and controller to the Upsy-Desky. It also updated as I raised and lowered the desk with the up/down buttons and presets. It also flashed when pressing the “M” button.

KiloHotl, 2026-01-13:
> Anyone still following this will be happy to hear that I've implemented and successfully tested WN17CM2/WN17CM3 in a fork of the esphome-standing-desk project which is what upsy-desky uses for its controller interface. Thanks to [@swoga](https://github.com/swoga) for his reverse engineering and Claude Opus 4.5 for making it quick enough for me to finally take off my todo list.
> 
> I don't have time to put it together into a PR until I've completed the rest of my project, which is to also add multiplexing support so that HA can raise/lower a collection of desks in unison - each of which has its own ESP32 networked wirelessly to a master ESP32 using ESP-NOW. Once that is done, I'll probably make separate PRs for the new controller and the multiplexing support. Expect those in 2-3 weeks.
> 
> Note that WN17CM2/WN17CM3 works differently than the other controllers and needs to send and receive control sequences via UART. Also, the desk controller operates at 5V while the ESP32 GPIO pins operate at 3.3V, so you will need a logic level converter if you want it to function reliably. I'm not sure if that's already handled by the upsy-desky hardware spec. If not, you may need to bring your own hardware (esp32, logic level stepper, ethernet port(s)) which should cost under $25 USD.

First of all, thanks for the effort. I've been waiting for this PR for a long time, but I don't know if this proyect should be with the upsy-desky due the comms and electronics are quite different.