tjhorner/upsy-desky #9: Questions on what I have seen so far...

jhspyhard, 2022-09-06:
I have just a few questions based on playing with the Upsy-Desky over the weekend:

1. Is it to be expected behavior that after setting artificial min / max height values in UD, that the desk's keypad will still allow raising and lowering outside of those limits? Is there any way to get the UD to force the keypad to respect those limits?
2. I'd like to experiment with setting up the webserver and API with HTTPS. Is SSH to the device something that can be turned on? Or maybe it's possible to specify the webserver cert / key values through the YAML config somehow? 
3. On that note, is there documentation on all of the available YAML config options which are available through this unit?
4. My Jarvis desk only has up and down controls, but I would love to be able to set some button presets using the UD and be able to recall them using the standard API method. Is there maybe an undocumented API method for setting those button presets on the UD itself if those memory locations don't exist in the desk's controller?

Super cool little device!  Thanks for a neat IOT gadget!

tjhorner, 2022-09-06:
1. Yes. I believe I can make it sync with the keypad — I'll look into it.
2. All of the options for the web server are available [here](https://esphome.io/components/web_server.html), but it doesn't look like it supports TLS. Personally I would use a reverse proxy that adds TLS.
3. The firmware is powered by ESPHome; all of its configuration options are documented [here](https://esphome.io/).
4. Not currently, but I can definitely add that! I'll target it for the next firmware update.

zeel01, 2022-09-25:
I was also wondering about the ability to store the presets (for Uplift). Modifying the stored data on the desk, be it the min/max height, the presets, or possibly the calibration values? All would be really cool. It would also be cool to be able to read those values out.

I was also a little surprised to learn that the min/max values are defined in the firmware rather than being pulled from/stored to the desk controller.

tjhorner, 2022-09-26:
> I was also a little surprised to learn that the min/max values are defined in the firmware rather than being pulled from/stored to the desk controller.

Unfortunately the desk does not report the min/max height to the keypad (and it's pretty finicky to set at runtime with ESPHome) so I'm not sure how feasible this is to change

zeel01, 2022-09-26:
I figured it was something like that.

tjhorner, 2022-10-02:
Closing this issue now. Let me know if you have any more questions @jhspyhard!

closed 2022-10-02 by tjhorner