tjhorner/upsy-desky #10: Conflicting ID: `status`

bachya, 2022-09-13:
I'm attempting to use [the default config](https://github.com/tjhorner/upsy-desky/blob/master/firmware/config.yaml) as part of a larger ESPHome config:

```yaml
---
packages:
  desk_config: "github://tjhorner/upsy-desky/firmware/config.yaml"
  device_base: !include common/device_base.yaml

substitutions:
  device_name: fully-jarvis-desk-controller
  device_friendly_name: Fully Jarvis Desk Controller
  device_static_ip: 192.168.10.100
  standing_desk_min_height: "25.3"
  standing_desk_max_height: "50.8"
```

When I attempt to compile this in ESPHome 2022.8.3, I get this error:

```
INFO Reading configuration /config/esphome/fully_jarvis_desk_controller.yaml...
Failed config

light.status_led: [source /config/esphome/.esphome/packages/111ab185/firmware/config.yaml:129]
  platform: status_led
  
  ID 'status' conflicts with the name of an esphome integration, please use another ID name.
  id: status
  pin: GPIO23
  restore_mode: ALWAYS_ON
```

https://github.com/tjhorner/upsy-desky/blob/849b4d889666f4864b80c6ea2a97799d757dea00/firmware/config.yaml#L128-L132

bachya, 2022-09-14:
FYI, I think this is the conflicting ESPHome integration referenced:

https://esphome.io/components/binary_sensor/status.html

tjhorner, 2022-09-19:
Fixed this, the ID should no longer conflict.

closed 2022-09-19 by tjhorner

bachya, 2022-09-19:
Thanks, TJ!

commit 66c7e075 (2022-09-19): Fix #10