tjhorner/upsy-desky #12: Simulating the up/down buttons.

zeel01, 2022-09-25:
I'm working on an integration to control the Upsy Desky using the REST API. And I would like to have up/down hold buttons just like the desk control panel has. But I don't see a direct way to do this.

Is there a command I can send that emulates these buttons? If not, is this something that could potentially be implemented? Or is there an issue making this impossible/problematic to do?

zeel01, 2022-09-26:
Here's what I've been working on: https://github.com/zeel01/upsy-sd-plugin

[image]

tjhorner, 2022-09-26:
That's a pretty neat integration! I will look into exposing an API endpoint for emulating the press or release of the up/down buttons. It shouldn't be too hard. Will update here when I can!

YangDaGod, 2025-02-24:
@tjhorner any update on this big guy?

VACIndustries, 2025-03-27:
This would be great!  I already bought two; having this ability I would probably pickup a few more (camera angles and such; it just makes life so easy!)