tjhorner/upsy-desky #13: Support for LoctekMotion/FlexiSpot desks

Jolanrensen, 2022-09-29:
Looks like most of the work is already done:
https://github.com/iMicknl/LoctekMotion_IoT
I think it would be awesome to have some sort of merge between these projects so that a clean all-in-one (no soldering) solution might also exist for Loctek desks.

tjhorner, 2022-10-01:
Definitely on the roadmap! Will need an adapter cable as noted in #5.

Jolanrensen, 2022-10-01:
Awesome! So same connector, different pinout. That's a bit more tricky than expected, but at least it's the same connector :)

Jolanrensen, 2022-10-13:
I ordered one! Order number #351096
If you already have access to an adapter for that desk for me to try, please let me know! Otherwise I'm also willing to wait for the upsy desky until you can ship it together with an adapter. That's fine by me :) let me know how I can be of help

Jolanrensen, 2022-11-03:
Upsy desky in the house! Now I just need an adapter :) (or a small guide on how to rewire the cables and make an adapter cable)

Jolanrensen, 2022-11-07:
I'll move my software findings here, so the adapter research (https://github.com/tjhorner/upsy-desky/issues/5) can happen in the right issue. Using the adapter described in that issue and my adapted version from LoctekMotion_Iot: https://github.com/Jolanrensen/LoctekMotion_IoT/tree/main 

I now have the following working: 
- up/down switch (requires virtual screen to be turned on)
- min/max height
- cover up/down/stop
- target height in cm
- 90% of the time smooth motion and no crashes anymore on the upsy desky

I did find that we probably cannot use the upsy desky templates atm, since those assume setting a pin on/off will control the desk, while LoctekMotion (and more) desks require uart streams instead. We could however maybe generalize that to a switch that should be turned on instead since that's how I got it to work. Also, the target height seems to work very accurately :), no +0.4 needed.

Now, getting the screen and buttons to work is next. I'm curious to see what I can do with those. Since the target height works so well from home assistant, I might connect the buttons directly to home assistant as well, allowing them to map to anything you'd like to happen... Don't know yet. The LocktekMotion_Iot project simply captures the signal from the buttons and activates a switch which then sends the signal to the desk again, but the preset buttons don't seem to do anything for me atm, so I'm not sure where those presets are physically saved.

Jolanrensen, 2022-11-08:
@tjhorner btw, is there a reason you use "return id(desk_height)->get_last_read() > x;" instead of "return id(desk_height).state > x;"? Is that for performance or for something else?

tjhorner, 2022-11-08:
@Jolanrensen It's so the automation is more responsive. The state is published less often than the height is read.

Jolanrensen, 2022-11-21:
Additionally fixed in my branch (https://github.com/Jolanrensen/LoctekMotion_IoT/tree/main)
  - Target height in cm now always lands on the right number. Even if it overshoots at first, it will go up/down till it hits the correct height exactly. (Can be quite comical sometimes)
  - Buttons 1, 2, 3 don't work passed through to my desk for some reason. However, you can now set 3 presets cm in Home Assistant (saved in EspHome for when Hass is offline) where the desk will move to
  
Need to decide what to do with M, A buttons. Maybe I'll recreate some sort of save-function.

Jolanrensen, 2022-11-23:
Pressing M and then 1, 2, or 3 saves the current desk height to the respective preset. Press M again (or any of the other buttons) to cancel.
Unfortunately, the screen doesn't display that you're saving something since the saving functionality of the desk itself doesn't seem to work for me in this setup

r14n, 2023-10-24:
What's the status of Flexispot support?  There's some decent looking ones on Amazon

Jolanrensen, 2023-10-25:
@r14n I've been using my [adapter](https://github.com/tjhorner/upsy-desky/issues/5) and [esphome file](https://github.com/Jolanrensen/LoctekMotion_IoT/blob/main/packages/esphome/hs01b-1_upsy_desky.yaml) successfully for a couple of months now. It's not perfect but it works.

r14n, 2023-10-26:
> @r14n I've been using my [adapter](https://github.com/tjhorner/upsy-desky/issues/5) and [esphome file](https://github.com/Jolanrensen/LoctekMotion_IoT/blob/main/packages/esphome/hs01b-1_upsy_desky.yaml) successfully for a couple of months now. It's not perfect but it works.

Do you think it would work with this one?  Bit hard to tell if the controller is RJ-45.  

https://www.amazon.com/gp/product/B0CHVQ5LCL/ref=ox_sc_act_title_1?smid=A3686AOLDPXXOD&th=1

[image]

Which desk model do you have btw?

Jolanrensen, 2023-10-26:
> Do you think it would work with this one? Bit hard to tell if the controller is RJ-45.
>
> https://www.amazon.com/gp/product/B0CHVQ5LCL/ref=ox_sc_act_title_1?smid=A3686AOLDPXXOD&th=1
>
> [image]
>
> Which [...]

That looks like a completely different controller. Probably RJ-45 since most desks are controlled using that, but we can't be 100% certain. Might also require a different adapter compared to mine. I've got a HS01B-1 controller.

r14n, 2023-10-26:
Are we able to identify this controller?  Searching for your HS01B-1 is not really turning up useful results (as far as desks I can purchase right now)

Keypad：
E2 Pro: 4 Presets Buttons with USB Ports
[image]

Jolanrensen, 2023-10-26:
> @Jolanrensen also, are you selling your adapter or willing to? Wishing for an "off the shelf" solution, as I'm familiar with homeassistant, but my electrical/wiring skills are very minimal. Seems like the upsy-desky isn't currently totally plug n play with flexispot. I'm also trying to keep my desk budget < $300 including additional "smart" hardware.

Haha I'm flattered, but my electrical skills aren't sell-worthy, plus I can't test it with your desk to see if it would work. Besides, my skills were minimal too before starting. I opted for the upsy-desky over a general ESP board since it already had a case and RJ45 ports. There are many toolless connectors that allow you to more easily make an adapter like mine, but nonetheless it will require some effort on your side. AFAIK there's no off-the-shelf solution for Flexispot desks.

Loic691, 2024-06-11:
Hi To follow this thread.
I just bought an E7 flexispot with this screen
A chance to get it work with Upsy ?
What kind of job we have to produce for flexipot compatibility ?
[image]
[image]
[image]

garymc, 2025-03-27:
Would be interested to know if anyone has made this work with Flexispot desks.

Ramiel99, 2025-05-16:
One more vote for Flexispot. I have E7Q (https://flexispot.co.uk/4-leg-standing-desk-e7q) 

[image]

[image]

[image]

L-e-c-o, 2025-08-09:
Great project, I was wondering if there were any updates on compatibility with Flexispot ?

KingOfSpades, 2025-10-30:
> Great project, I was wondering if there were any updates on compatibility with Flexispot ?

Wondering the same!