tjhorner/upsy-desky #14: Jarvis Desk V2 Hardware Target Height requires two commands

Trinnik, 2022-10-04:
I recently received my order for the V2 hardware to hookup to my Fully Jarvis desk (fully assembled). This specific desk uses the uplift decoder. I have the desk working and hooked up to Home Assistant.

When attempting to make the desk go to it's maximum or minimum height, it always requires two commands to be sent. For instance, the max height of my desk is 49.2 inches, however, when I tell the desk to go to 49.2 it will always stop at 43.2. If I send the command again to go to 49.2 it will complete it's travel and reach max height of 49.2. The same thing happens on the way down, my minimum is 30.0 inches, however, it will always stop around 35.2.

A separate issue I noticed, I have to set my limits slightly above and below the reported min and max heights to actually reach them. Otherwise they always stop .3 inches before each value (using the hardware button will get me all the way).

tjhorner, 2022-10-06:
Thanks for the information! I will look into a fix for this.

SnooSnooKachu, 2022-10-06:
I'm experiencing the same thing. I find the further I ask the desk to move in one jump, the further off the end result is. For example, when I ask it to move from 27.7" to 47.2", it stops moving at 41.6". But if I ask it to go from 27.7" to 33.1", it stops at 33.0".

Trinnik, 2022-10-07:
Seems like the timeout in template.yaml needs to be increased from 10 seconds to 20 seconds. Timing my desk it takes 15 seconds to move it's full range, 20 should be enough to cover most cases. Changing this on my device solves this issue.

Also seems like there is a related issue already in that repository. tjhorner/esphome-standing-desk#1

tjhorner, 2022-10-08:
@Trinnik I’m glad that change works for you, I will go ahead and update the timeout to 20 seconds in the stock firmware so others don’t encounter this in the future.

@SnooSnooKachu could you also verify that increasing the timeout fixes the issue?

Thanks all!

SnooSnooKachu, 2022-10-11:
@tjhorner, where can I get this firmware? The latest I see in [this Github](https://github.com/tjhorner/upsy-desky/releases) is v0.1.1. I flashed that onto the device, but it didn't fix anything. I also tried the web flash [here](https://shop.horner.tj/things/upsy-desky/setup/stock), but that just reflashed the v0.1.1 I already had.

tjhorner, 2022-10-11:
@SnooSnooKachu I just released version 0.1.2 that includes a fix for the issue. I had assumed you had your own ESPHome config, my bad! You can flash it from [here](https://shop.horner.tj/things/upsy-desky/setup/stock) now.

SnooSnooKachu, 2022-10-11:
@tjhorner I was able to flash the new firmware and it works like a charm!

tjhorner, 2022-10-11:
Awesome, thanks for confirming. I'll close this issue now :)

closed 2022-10-11 by tjhorner