tjhorner/upsy-desky #15: IKEA UPPSPEL support

murraythegoz, 2022-10-15:
Hi,

asking for support for the IKEA's UPPSEL
controller seems to be a rebranded JCB36N2 
(https://www.jiecang.com/JCB36N2-pd45762211.html)

Works partially.

I set both uplift and jarvis template.
All 4 desk preset works (both from web page and via MQTT)
Current height is not handled.

How can I help debugging?

thank you 

m.

murraythegoz, 2022-10-15:
A small update.
desk height is handled, but there is an offset.

Lowest height: 72 cm

height #1: 108 cm
reported height #1: 26 cm

height #2: 72 cm
reported height #2: 25.9 cm

height #3: 120 cm
reported height #3: 26 cm

murraythegoz, 2022-10-15:
[image]

murraythegoz, 2022-10-15:
[image]

murraythegoz, 2022-10-15:
[image]

tjhorner, 2022-10-16:
Thanks for the detailed report! I can make a separate decoder which offsets the height by -72 cm and send that to you for testing. I'll let you know when that is available.

murraythegoz, 2022-10-17:
Thank you,

please let me know if in the meanwhile can help somehow (e.g. how to check whether the problem is due not only to offset or also different way of handling/decoding actual height)

murraythegoz, 2022-11-04:
Hi any update?
May I help somehow?

thanks

tjhorner, 2022-11-05:
Hi @murraythegoz, sorry for the delay. A lot on my plate right now! I should have a firmware for you to test this weekend.

nicholasgk, 2022-12-16:
I had the same issue with my [Omnidesk Pro](https://theomnidesk.com.au/pages/2020). I found that the file [uplift_decoder.cpp](https://github.com/tjhorner/esphome-standing-desk/blob/master/components/standing_desk_height/uplift_decoder.cpp) on line 27, was only looking for a value of 0 or 1. The desk was returning values up to 4. I think it is due to mm vs inches. I changed this one line to
`if (b == 0x00 || b == 0x01 || b == 0x02 || b == 0x03 || b == 0x04) {
`
and it started returning the correct height values.

I am not a C++ programmer, so there probably is a better way to write this. Hope this helps.

murraythegoz, 2022-12-16:
> I had the same issue with my [Omnidesk Pro](https://theomnidesk.com.au/pages/2020). I found that the file [...]

Yup. this made the trick!

I'd instantiate a new variant with the suggested change, to avoid scramble working variants.

Thank you!

closed 2023-02-01 by tjhorner

pedrams1, 2023-02-05:
Tested this and confirming it resolves the issue reported here - https://github.com/tjhorner/wifi-desk-controller/discussions/18

Thank you so much for this!!! The desk is working perfectly now :)

tjhorner, 2023-02-06:
@pedrams1 could you provide the model number for your desk's control box? I'd like to add it to the list of supported desks in the documentation. Thanks!