tjhorner/upsy-desky #16: Consider adding fast_connect to WiFi options

TheRealAvise, 2022-10-23:
Maybe consider adding fast_connect option under WiFi, otherwise it can't connect to hidden SSID's.

tjhorner, 2022-10-24:
I have actually tried this out before; ESPHome requires that a WiFi network is specified in the config for this option to be available. I'm unsure if they have changed this in recent versions, but I'll see if this is now possible. If it isn't, I'll raise an issue on ESPHome's side and maybe submit a PR to enable this functionality without specifying a network in the config

TheRealAvise, 2022-10-24:
Ups, did not know that.

Maybe then just add to the wiki, as something to try, if WiFi SSID is hidden.

tjhorner, 2022-10-24:
Yep, good idea! I'll add that to the troubleshooting doc

tjhorner, 2023-05-27:
Added to the Getting Started docs

closed 2023-05-27 by tjhorner