tjhorner/upsy-desky #19: ESPHome Bluetooth Proxy Capabilities

topperge, 2022-11-04:
Home Assistant added Bluetooth proxy support back in September to extend the reach of Bluetooth across your house. Given that the ESP32 is the main board for this integration, it would be great to have the option to enable that on the Upsy Desky. I'd like to set up some integrations to help force me to stand more during the day or control lights if I'm in my office and have a meeting starting or ending. Since the Upsy is right under me, there isn't a much better option to prove if I'm sitting there or not. I've never worked with the ESP32 but I think I have a file that'll drop right in.
[config.txt](https://github.com/tjhorner/upsy-desky/files/9935098/config.txt)

tjhorner, 2022-11-04:
Sure, great idea! I will run some tests to make sure it doesn't affect performance or the size of the stock firmware too much, and I'll update accordingly.

JayNewstrom, 2022-11-04:
I've been running a modified version of the firmware in the repo that also uses `bluetooth_proxy:`, and so far it's been stable!

jhoff80, 2023-03-11:
Any updates on this?  Love the Upsy Desky already, but it would be helpful to have a Bluetooth proxy in the office without needing to add a second device.

tjhorner, 2023-03-12:
I have added the `bluetooth_proxy` component by default starting in 0.4.1. You can learn how to update your firmware [here](https://upsy-desky.tjhorner.dev/docs/firmware-updates/).

Let me know how it works out!

closed 2023-03-12 by tjhorner