tjhorner/upsy-desky #20: Accuracy/Granularity of Height

ohadeytan, 2022-11-12:
Hi, thanks for all the development :) 

I'm using the rest API with Uplift desk. 

When issuing a GET request, I get an accurate 0.1-inch precision as on the physical panel.
But, when I PUT some height, it often goes to a close-by position, only about 0.5-inch precision. Following PUT requests to close positions not moving the table at all. 

Is this the expected behavior? Is it configurable or could easily be fixed by changing the firmware?

tjhorner, 2022-11-14:
Yes, setting the desk height via that slider/PUT request is not going to be exact. It basically emulates pressing the up or down button until the desk _almost_ reaches the height. It's different from, e.g. pressing one of the preset buttons, where it slows down when getting close.

Unfortunately Uplift desks do not accept commands like "go to this specific height" so the best we can do is approximate it. Something I _do_ recommend is using the presets feature — you can set and get them from the Upsy Desky and they are stored in the control box (same as the keypad, if yours supports presets). Then it will utilize the desk's built-in soft-stop when heading to a preset, and you can use the HTTP API to recall them as well.

ohadeytan, 2022-11-14:
Understood. I will try the presets. 
Out of curiosity, can you please direct me to the place in the code where this approximation is happening?

tjhorner, 2022-11-15:
Sure: https://github.com/tjhorner/esphome-standing-desk/blob/eee7657ebd62d96d499ef8457b24b45c1e0dc7b2/configs/template.yaml#L45-L62

ohadeytan, 2022-11-15:
Great, thanks!

closed 2022-11-15 by ohadeytan