tjhorner/upsy-desky #22: ROL ERGO Support

magnussp, 2022-12-16:
I'm trying to check see if I can get Upsy-Desky to work with a ROL ERGO GF-470 (Swedish manufacturer of standup desks, I do believe they have sold some controllers to IKEA as well). This is what I have found so far.

This is a picture of the circuit board of the controller box which the original keypad for the desk connects to.
[image]

This is how the inside of the keypad looks like
[image]

What we can conclude from this is that according to the [Desk Compatability](https://github.com/tjhorner/wifi-desk-controller/wiki/Desk-Compatibility) the ROL ERGO desk is not compatible of the bat by simply pluging in a ethernet cable.

But seeing that it had 5V, RX and TX pins made me hopeful so I got a RJ45 breakoutboard and tried to make a cable that would match the pinout out the ROL ERGO controller with what Upsy-Desky needs. However I have had no luck so far.

What I noted was that the 5V from the ROL ERGO controller drops to bellow 1V when I connect Upsy-Desky so I assume the 5V from the controller is powerful enough to handle data signals and not to drive the Upsy-Desky. To get around this and troubleshoot I simply attached usb power to Upsy-Desky while everything was connected to the ROL ERGO controller.

Worth noting could be that from the picture of inside the keypad the desk will move up if red+green wire is shorted and it will move down if the white+green wire is shorted.

Do you think there is a chance to get this desk working with Upsy-Desky, and how could I then assist in the troubleshooting?

magnussp, 2023-03-06:
@tjhorner tell me if there is something I can assist with to see if it would even be possible to add support for this desk.

tjhorner, 2023-03-07:
So sorry I missed this! I will review what you've posted tomorrow and let you know.

dcobbley, 2023-03-23:
I have the ROL ECS+ controller and have not had luck yet cracking the exact UART communication settings yet. My RJ45 pinouts are unlabeled so it has taken me some time to trace the PCB and figure out which lines represent RX,TX, etc.

Please take a photo of your entire PCB (top & bottom), as high resolution as you can. It will help me compare against my setup (and hopefully provide helpful info back to you) which uses the [PIC18F26K22](https://www.microchip.com/en-us/product/PIC18F26K22) in the main controller, and the [PIC16F1938](https://www.microchip.com/en-us/product/PIC16F1938) in the handset.

Reading through the manual of the ROL ERGO GF-470, it looks like all the handset does is provide an up/down button to manually control height as well as providing a way to set two height limits when both up & down are pressed simultaneously.

I could be way off here, but I wonder if you could simplify making your controller 'smart' by simply not using the UART portion of upsy-desky and only rely upon the GPIO's to press the up/down button for x seconds at which point the desk controller will automatically stop the desk at the originally programmed height. You could even prototype this with any ESP dev board if you haven't yet purchased an Upsy-Desky board.

magnussp, 2023-04-20:
@dcobbley sry for the late reply. Here are some pics of the entire board in my ROL ERGO desk.

[image]
[image]

Your observation from the manual is correct. I have been thinking of this solution as well to just connect GPIO to the original controller and then set min and max limits in the oem way for the desk and control it this way. But would be cool to get that working through Upsy-Desky 🙂

CM000n, 2024-01-10:
There are now also some instructions for assembling your own controller for ROL ERGO tables using an RJ45 extension cable and an ESP device: https://github.com/jensihnow/smarthome/blob/main/smartdesk/README.md
These ROL ERGO controllers are also used for IKEA Rodulf tables.
Compatibility with Upsy Desky should therefore be "theoretically" possible. 😉

Xavantex, 2024-12-21:
As it seems ROL ERGO just have binary on/off for up and down, is upsy-desky even compatible with some manual labor?