tjhorner/upsy-desky #24: Desk Height not being reported

p-rintz, 2023-01-25:
Hi there,

thank you for the great product, first.

Ive been using it without a problem since it arrived, sadly, since updating all my ESPhome controllers yesterday, the upsy-desky stopped reporting the desk height. It still works otherwise, but it only shows N/A for the height.

Rolling back to the firmware uploaded to Github has not helped this.

Please do not hesitate to tell me what further information you need. Thank you!

[image]

tjhorner, 2023-01-26:
Is the decoder set correctly? I wonder if the update has reset it back to the default.

Is there anything interesting in the logs? Errors or such would be helpful

Try using this tool to entirely reflash and see if it works: https://shop.horner.tj/things/upsy-desky/setup/stock

p-rintz, 2023-01-26:
I already tried setting the decoder and the units to uplift/inches and back again, and have tried it again just now, sadly, that hasn't helped.

I will try reflashing it later today.

p-rintz, 2023-01-26:
Well, I couldnt wait. So I tried reflashing it just now via the browser, but it hasnt helped either, sadly.

Should I try flashing the debug firmware that you have on the release page?

dominiclesinski, 2023-01-27:
just gonna jump in here rather than opening a separate issue because I also have a problem with desk height reporting. I changed it to cm but it only changed the unit rather than the actual number so I now get 28cm rather than 28in, while in reality it should be  71cm :) any advice? ty!

p-rintz, 2023-01-27:
> just gonna jump in here rather than opening a separate issue because I also have a problem with desk height reporting. I changed it to cm but it only changed the unit rather than the actual number so I now get 28cm rather than 28in, while in reality it should be 71cm :) any advice? ty!

Restart the upsy-desky. It should work afterwards.

dominiclesinski, 2023-01-27:
thanks for trying to help, tried that already several times with the same result...
firmware version is : 2022.10.1

junoatwork, 2023-01-27:
Came here after noticing the same "NA" reading. The keypad on my desk says "jarvis", and when I bought the desk the brand name was under "jarvis", so naturally I chose the "jarvis" decoder in the upsy-desky web interface. 

However, after further investigating the control unit under my desk (googling the model number model: `jcb36n2a`) turns out to be "uplift" - changing to that decoder works, and now the number reported by the sensor matches what's shown on the lcd on the original keypad.

Posting in case it helps anyone else!

p-rintz, 2023-01-27:
@junoatwork thank you. While this wasnt the issue I had (as it worked previously as well, with the jarvis decoder), your comment helped me in fixing my issue.
I hadn't looked at the control unit since installing the upsy-desky (because it just works so well), but for some reason the control unit itself seemed to have "bugged" out after the update and wasn't displaying the height itself anymore either.

After unplugging and re-plugging the control unit into the upsy-desky, it started working again.

Not entirely sure how it happened or if this can happen regularly. 

While my issue obviously is "solved", Ill keep this open for now for @tjhorner to pipe in, in regards to this potentially being some bug with the upsy-desky.

p-rintz, 2023-01-27:
> just gonna jump in here rather than opening a separate issue because I also have a problem with desk height reporting. I changed it to cm but it only changed the unit rather than the actual number so I now get 28cm rather than 28in, while in reality it should be 71cm :) any advice? ty!

I also just re-read your comment. Sorry, I misunderstood you earlier. I had a similar "display" issue, but that isnt what you meant, if I read your comment correctly now. You probably need to change the decoder. Uplift --> Jarvis or the other way around.

tjhorner, 2023-01-28:
I'm glad you got it fixed @p-rintz, and thanks for posting your solution! I will go ahead and add this to the troubleshooting documentation in case others encounter something similar.

p-rintz, 2023-01-28:
If you dont think this is something that could have been triggered by the upsy-desky, feel free to close this issue @tjhorner 

Thank you again for the great product. :+1:

closed 2023-02-07 by tjhorner