tjhorner/upsy-desky #29: esphome MQTT support

junosuarez, 2023-03-29:
Thanks for upsy-desky! 

I'm currently running it on my network, and I have a separate server polling the http endpoint and publishing the data to various other data stores, including MQTT. I'm a complete novice with esphome. I recently found that they have builtin MQTT support, which would be perfect for my needs. 

Would this be useful to anyone else? @tjhorner, would you accept a PR adding [MQTT client](https://esphome.io/components/mqtt.html) to publish desk height messages? If so, I'm happy to do some research and learn my way through implementing it.

tjhorner, 2023-03-29:
MQTT support would be a great addition, and a PR for this would definitely be welcome. You could include the config in a new file in the `firmware/addons` directory, and provide substitutions for the broker, username, and password so that others can include it in their configuration as well.

If you need anything from me or have any questions about the existing configuration, let me know. Thanks!

junosuarez, 2023-04-17:
Update: I still want this and intend to implement it - but I got bogged down trying to configure the ESPHome dashboard following these directions https://upsy-desky.tjhorner.dev/docs/firmware-updates/  (this is not an ask for individual tech support). 

I'll give it another go when I have time. This is low prio for me, so if anyone else wants this more than me, feel free to pick this up from me :) Otherwise, I'll provide updates as I go, but I expect it to take a while.