tjhorner/upsy-desky #30: Support for child lock / control lock feature

nperanzi, 2023-04-05:
Some desks (e.g., Uplift V2) have a child lock feature. Enabling child lock via the keypad also prevents upsy-desky from controlling the desk, so this feature appears to be enabled on the control box itself, rather than the keypad.

Two enhancement requests related to this feature:

- Support for detecting if child lock is enabled or not
- Support for changing the status of the child lock (lock or unlock the desk)

tjhorner, 2023-06-28:
Hi, thanks for the suggestion, and so sorry I missed this. I looked into this and the only way to trigger the keypad lock is to actually emulate holding the "M" button for the time it takes to enable or disable it, since the keypad does not send commands with serial (or through any other means). This isn't ideal, but it should be possible.

Detection should be possible, and would be easy to expose via a `binary_sensor`. I will need to do some reverse engineering to see what packet the control box sends while the height is locked.

jfroy, 2024-01-03:
As the happy father of a one year old, is there anything I can do to help? I've done some dabbling with esphome, but I don't know the specifics of the Uplift controller.

jhemak, 2024-06-11:
+1 for this.  Thank you!

alexives, 2024-08-30:
As another parent of a small child, I decided to unplug my physical controller and use [one of these](https://www.thesmartesthouse.com/collections/zooz/products/zooz-800-series-z-wave-long-range-wall-remote-zen37-800lr-battery-powered) - hooked up to the esphome device.

Not exactly ideal, but it keeps my toddler from adjusting it while I'm sitting/standing at it working 😅