tjhorner/upsy-desky #31: Unable to update to ESPHome version to `2023.5.1` or have it connect to read the logs

nreymundo, 2023-05-18:
I'm facing an issue when trying to update the device using the ESPHome addon in Home Assistant. 

Here's the error message I get on the ESPHome console: 
```
/config/esphome/.esphome/packages/663b17eb/firmware/addons/runtime-config.yaml: In lambda function:
/config/esphome/.esphome/packages/663b17eb/firmware/addons/runtime-config.yaml:14:45: error: no matching function for call to 'esphome::standing_desk_height::StandingDeskHeightSensor::set_unit_of_measurement(std::__cxx11::string&)'
           id(desk_height)->set_unit_of_measurement(x);
                                             ^
In file included from src/esphome/components/light/light_state.h:4,
                 from src/esphome/core/controller.h:11,
                 from src/esphome/components/api/api_server.h:4,
                 from src/esphome/components/api/api_connection.h:6,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/core/entity_base.h:82:8: note: candidate: 'void esphome::EntityBase_UnitOfMeasurement::set_unit_of_measurement(const char*)'
   void set_unit_of_measurement(const char *unit_of_measurement);
        ^~~~~~~~~~~~~~~~~~~~~~~
src/esphome/core/entity_base.h:82:8: note:   no known conversion for argument 1 from 'std::__cxx11::string' {aka 'std::__cxx11::basic_string<char>'} to 'const char*'
/config/esphome/.esphome/packages/663b17eb/firmware/addons/runtime-config.yaml:15:59: error: no matching function for call to 'esphome::number::NumberTraits::set_unit_of_measurement(std::__cxx11::string&)'
           id(target_desk_height)->traits.set_unit_of_measurement(x);
                                                           ^
In file included from src/esphome/components/light/light_state.h:4,
                 from src/esphome/core/controller.h:11,
                 from src/esphome/components/api/api_server.h:4,
                 from src/esphome/components/api/api_connection.h:6,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/core/entity_base.h:82:8: note: candidate: 'void esphome::EntityBase_UnitOfMeasurement::set_unit_of_measurement(const char*)'
   void set_unit_of_measurement(const char *unit_of_measurement);
        ^~~~~~~~~~~~~~~~~~~~~~~
src/esphome/core/entity_base.h:82:8: note:   no known conversion for argument 1 from 'std::__cxx11::string' {aka 'std::__cxx11::basic_string<char>'} to 'const char*'
Compiling /data/upsy-desky-58032c/.pioenvs/upsy-desky-58032c/lib69d/AsyncTCP-esphome/AsyncTCP.cpp.o
*** [/data/upsy-desky-58032c/.pioenvs/upsy-desky-58032c/src/main.cpp.o] Error 1
========================= [FAILED] Took 84.68 seconds =======================
``` 

At first I thought it could be some incompatibility issue or some other local problem so I removed it from ESPHome, downloaded FW version 1.0 from this repo and flashed it again. When adopting it back the device gets properly added to Home Assistant but I run into the same error compiling the update. 

Also tried a fresh factory reset. Same issue. 

On top of that ESPHome seems to be unable to connect to it to read the logs. 

This is what I get from the ESPHome site when I click on the `Logs` option for the Upsy Desky. 
```
INFO Reading configuration /config/esphome/upsy-desky-58032c.yaml...
INFO Starting log output from upsy-desky-58032c.local using esphome API
WARNING upsy-desky-58032c.local: Connection error occurred: [Errno 104] Connection reset by peer
WARNING Can't connect to ESPHome API for upsy-desky-58032c.local: Connection isn't established yet (ConnectionState.CLOSED)
INFO Trying to connect to upsy-desky-58032c.local in the background
WARNING upsy-desky-58032c.local: Connection error occurred: [Errno 104] Connection reset by peer
WARNING upsy-desky-58032c.local: Connection error occurred: [Errno 104] Connection reset by peer
WARNING upsy-desky-58032c.local: Connection error occurred: [Errno 104] Connection reset by peer
WARNING upsy-desky-58032c.local: Connection error occurred: [Errno 104] Connection reset by peer
```

And this is the counterpart, from the logs in the Upsy Desky captive portal while ESPHome is trying to connect
```
16:28:19	[W]	[ota:393]	Remote closed connection
16:28:19	[W]	[ota:154]	Reading magic bytes failed!
16:28:43	[W]	[api.connection:085]	::FFFF:192.168.20.10: Reading failed: BAD_INDICATOR errno=11
16:28:46	[W]	[api.connection:085]	::FFFF:192.168.20.10: Reading failed: BAD_INDICATOR errno=11
16:28:49	[W]	[api.connection:085]	::FFFF:192.168.20.10: Reading failed: BAD_INDICATOR errno=11
16:28:56	[W]	[api.connection:085]	::FFFF:192.168.20.10: Reading failed: BAD_INDICATOR errno=11
16:29:06	[W]	[api.connection:085]	::FFFF:192.168.20.10: Reading failed: BAD_INDICATOR errno=11
```

tjhorner, 2023-05-18:
It looks like ESPHome has made changes to their internal API that are incompatible with the underlying [custom component](https://github.com/tjhorner/esphome-standing-desk), so I will need to make some changes there. I'll update the thread here when those changes are complete so you can re-compile on ESPHome 2023.5.

Regarding your connection issues:

- Does `upsy-desky-58032c.local` resolve to the correct IP?
- Once you have factory reset, you will need to remove any references to encryption keys (in your YAML and Home Assistant integration) since the stock firmware does not use one. That's probably why you are seeing the `Reading failed: BAD_INDICATOR` message in the logs.

tjhorner, 2023-05-18:
Ok, the issue was in the `runtime-config` addon — I just fixed it, please try compiling again. You might need to clean the build files so that it grabs the latest version of the package from the repo (or target the commit `274ec9ac9464bb23481422de32094e18a66e6d3f` in the YAML).

nreymundo, 2023-05-18:
Hey! Thanks for the quick reply. 

So some... progress?. Targeting `1.0.1` worked perfectly and it built but now it seems stuck in a connect/disconnect loop?. I tried another factory reset as well as removing it both from ESPHome as well as from the Home Assistant integration. 

Putting some screenshots here. Am I missing anything still?. 

ESPHome when trying to get the logs: 
[image]

Upsy Desky captive portal:
[image]

Home Assistant integration screen:
[image]

nreymundo, 2023-05-18:
Reverting to `1.0.0` fixed the issue in the screenshots above and allows HA to connect to the device again.

tjhorner, 2023-05-18:
That's strange. Can I see your entire YAML config?

nreymundo, 2023-05-18:
Totally, pretty bog-standard really. 

I reverted by flashing straight from the device's captive portal as switching the YAML to target `1.0.0` I assumed would have ended in a compilation error. That's also why in the YAML below you still see `1.0.1`. 

```
substitutions:
  name: upsy-desky-58032c
  friendly_name: office-desk-upsy-desky
packages:
  tj_horner.upsy_desky: github://tjhorner/upsy-desky/firmware/stock.yaml@v1.0.1
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: <rando encryption key here>

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

tjhorner, 2023-05-18:
Gotcha, makes sense. Can you try these steps?

1. Compile and flash your YAML targeting 1.0.1
2. Remove the existing ESPHome integration from Home Assistant
3. Add the ESPHome integration back

Since the connection works with the stock firmware, and that doesn't have an encryption key set, I think Home Assistant is attempting to connect without one, which means it never successfully connects, resulting in all the failures in your log. Re-adding the integration with your encryption key should solve this.

nreymundo, 2023-05-18:
Already tried that before with no luck, I guess there's a part of the config sticking around somewhere? It's weird honestly. 

Here are the steps I tried:
1. Remove the device from the integrations. 
2. Remove the device from ESP Home. 
3. Restart Home Assistant (just to be safe). 
4. Re-Adopt the device. 
5. Flash latest version - All great here. 
6. Add it to the integrations. 

I'm still getting the failure to connect message but poking around the UI or the file system I can't find a place where to edit the key or remove past configs?.  Heck, I even re-added it with a different name just to be sure and still the same issue. 

If I remove the encryption key and flash again everything goes perfectly and connects great ¯\_(ツ)_/¯. 

Thanks for the quick assistance though!.

tjhorner, 2023-05-21:
Thanks for the additional information. I'll do some more testing on my end and get back to you.

Since the original problem you opened this issue for is resolved, I'll close this and we can move discussion to #32.

closed 2023-05-21 by tjhorner