tjhorner/upsy-desky #34: 1.1.1 firmware, getting connection errors from Home Assistant

dhuddle, 2023-05-27:
[image]

I can't seem to connect to view logs and these errors and the device is unavailable in home assistant. I can connect to the device via the .local address and I've verified the latest firmware.

[image]

tjhorner, 2023-05-27:
Hi, can I have more information about how you are updating to 1.1.1? Was it via an OTA, or do you manage your own ESPHome configuration? If it's the latter, can you please share the YAML (with encryption keys, passwords, etc. redacted)?

This seems reminiscent of #31 which occurred a few versions ago.

dhuddle, 2023-05-27:
I have 2; I updated both to 1.1.1 via OTA (uploaded via each device’s web interface.)

They are both adopted in ESPHome, but I can’t compile the firmware that way, so I had to do it directly.

tjhorner, 2023-05-28:
Alright, thanks for that info. Can you first try restarting Home Asssistant completely, and if that doesn't work, try deleting and re-adding the ESPHome integration for one of your devices in Home Asssistant?