tjhorner/upsy-desky #36: Zigbee/Thread version

35gh, 2023-06-15:
Hi,

I would love to get a Zigbee/Thread version of the Upsy-Desky. Would that require a new hardware revision or can the support be added via a firmware update?

Thanks!

tjhorner, 2023-06-15:
This would require a hardware revision and a pretty significant firmware overhaul, so while I also love Zigbee and Thread I don't see this happening for a while, especially since this is not a battery-powered device so the benefits would be minimal.

But it's something I would love to work on in the future when I have more free time, so I'll keep this issue open.

Skaronator, 2023-11-01:
This is basically blocked by https://github.com/esphome/feature-requests/issues/1397

Looks like some progress happened, but not much.