tjhorner/upsy-desky #41: Fatal errors in ESPHome 2023.12.0b1

NewsGuyTor, 2023-12-15:
### Troubleshooting Guide

- [X] I have followed the steps in the troubleshooting guide and my issue is either not listed or is not solved.

### Description

With ESPHome 2023.12.0b1, Upsy Desky can't be controlled/updated anymore since the same pins are used twice.

### Logs

```plain text
INFO ESPHome 2023.12.0b1
INFO Reading configuration /config/esphome/upsy-desky-9fb5cc.yaml...
Failed config

output.gpio: [source <unicode string>:24]
  
  Pin 21 is used in multiple places.
  platform: gpio
  inverted: True
  id: standing_desk_up_pin
  pin: 
    number: 21
    mode: 
      output: True
      input: False
      open_drain: False
      pullup: False
      pulldown: False
    inverted: False
    ignore_strapping_warning: False
    drive_strength: 20.0
output.gpio: [source <unicode string>:29]
  
  Pin 22 is used in multiple places.
  platform: gpio
  inverted: True
  id: standing_desk_down_pin
  pin: 
    number: 22
    mode: 
      output: True
      input: False
      open_drain: False
      pullup: False
      pulldown: False
    inverted: False
    ignore_strapping_warning: False
    drive_strength: 20.0
output.gpio: [source <unicode string>:2]
  
  Pin 21 is used in multiple places.
  platform: gpio
  inverted: True
  id: button_bit1
  pin: 
    number: 21
    mode: 
      output: True
      input: False
      open_drain: False
      pullup: False
      pulldown: False
    inverted: False
    ignore_strapping_warning: False
    drive_strength: 20.0
output.gpio: [source <unicode string>:7]
  
  Pin 22 is used in multiple places.
  platform: gpio
  inverted: True
  id: button_bit2
  pin: 
    number: 22
    mode: 
      output: True
      input: False
      open_drain: False
      pullup: False
      pulldown: False
    inverted: False
    ignore_strapping_warning: False
    drive_strength: 20.0
```

### Which version of the Upsy Desky firmware config are you running?

1.2.0

### How do you manage your Upsy Desky's updates?

ESPHome YAML Config

### If you're using an ESPHome YAML config, which version of ESPHome are you running?

2023.12.0b1

### ESPHome Config

```YAML
substitutions:
  name: upsy-desky-9fb5cc
packages:
  tj_horner.upsy_desky: github://tjhorner/upsy-desky/firmware/config.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false
api:
  encryption:
    key: dfsjkfdfjsdjnkfsjhkd

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

tjhorner, 2023-12-16:
Thank you for the heads up. I'll aim to fix this before ESPHome 2023.12 is out of beta, but I'm going to be on vacation for the next few weeks so it's unlikely.

In the meantime please use the latest stable version of 2023.11.

tjhorner, 2023-12-16:
After a brief investigation it looks like this change was introduced in this PR: https://github.com/esphome/esphome/pull/5666

There is an option to suppress this error for individual pins, so it will be easier to fix than originally thought. I can probably come up with a fix this weekend.

Though, I haven't really thought that much about the consequences of pins having multiple uses, so that's probably worth revisiting anyway. (I've always been aware that it could be a Bad Idea, but nothing has really gone wrong because of it yet.)

tjhorner, 2023-12-17:
@NewsGuyTor Can you change your config to reference the package `github://tjhorner/upsy-desky/firmware/stock.yaml@esphome-2023.12`? You should be able to compile successfully with that. I'll publish a new release with those changes once ESPHome 2023.12 is out of beta.

NewsGuyTor, 2023-12-17:
Seems to work fine! Updated to 2023.12b2 without issue, and tested controlling the desk through HA still works great.

tjhorner, 2023-12-21:
Version 2.0.0 released with these changes.

closed 2023-12-21 by tjhorner