tjhorner/upsy-desky #43: Does it support WP-CB01 Control box?

oeloo, 2024-02-29:
Hi, I have a desk with a WP-CB01 Control box, is upsy-desky able to control it?
[image]

tjhorner, 2024-03-01:
This model number doesn't appear in the [compatibility list](https://upsy-desky.tjhorner.dev/docs/reference/compatibility/) so I can't guarantee that it will. However, please check out the [compatibility assessment guide](https://upsy-desky.tjhorner.dev/docs/reference/compatibility/assessing-compatibility/) — it includes useful information on how to determine if it may be compatible

oeloo, 2024-03-01:
I have looked at the guide and checked my Control box: the port that communicate with the keypad has 10 pins, so it is slightly different from an Ethernet jack. If I read correctly this means it means there is no chance it will work 😟

tjhorner, 2024-03-01:
That's correct, if it uses either of the outer-most of the 10 pins then I'm afraid it won't work 😔

oeloo, 2024-03-01:
The outer-most are fired on the port end, I don't know if they are wired inside the remote control PCB (as there is no screws or anything to open it safely without destroying it) but I guess it is otherwise they would have chosen a RJ45 plug I suppose.
[image]
[image]

tomas1808, 2025-05-09:
Hey @tjhorner just received my Upsy Desky! I YOLO'd it hoping that my desk would be compatible but it wasn't lol. I have the same controller as OP with the 10-pin cable. 

My idea is to change the wiring to make it work. I will remove the 10-pin cable from the keypad and replace it with a standard 8-pin RJ45 with the correct remapping of pins. This should be able to connect correctly to Upsy Desky. Then I'll use the 10-pin cable to make a "8-pin to 10-pin" cable so that I can hook Upsy Desky to the controller box without having to mod the controller box itself. 

I have already identified +5V and GND pins (pins 9 and 6 respectively on the 10-pin cable). My only question to you @tjhorner is: If I correctly set up the +5V and GND pins to be compatible with Uspy Desky, can the remaining data pins be detected by Upsy Desky and "remapped" by software? I really don't want to take out my logic analyzers to figure out the rest of the pins lol. 

Thanks!