tjhorner/upsy-desky #44: Preset Button 2-4 are only moving the desk a small bit but Preset 1 works as expected

plaetzchen, 2024-03-25:
### Troubleshooting Guide

- [X] I have followed the steps in the troubleshooting guide and my issue is either not listed or is not solved.

### Description

Installed my Upsy Desky to my aeris Smart Desk and the preset button 1 works as expected, setting the height of the desk to the stored value after clicking it. However, clicking on the presets buttons 2-4 does move the desk into the right direction but only for a centimetre and it stops again. The buttons of the desk work as expected, the set desired height works as expected.

### Which version of the Upsy Desky firmware config are you running?

1.1.1

### How do you manage your Upsy Desky's updates?

OTA Updates

### If you're using an ESPHome YAML config, which version of ESPHome are you running?

2023.5.4

plaetzchen, 2024-04-03:
@tjhorner Thanks for the quick response, the model on the control box is JCB36N2AG-230 from Zhejiang Jiecang Linear Motion Technology Co.

Kerryliu, 2024-10-22:
I'm also seeing a similar behavior with my jarvis standing desk (FULLYCB2C-A).