tjhorner/upsy-desky #45: Clarify BOM for manufacturing PCB

Crazyphil, 2024-04-01:
### Troubleshooting Guide

- [X] I have followed the steps in the troubleshooting guide and my issue is either not listed or is not solved.

### Description

I'm trying to assemble a PCB order directly from a PCB manufacturer (not because I don't like you earning some money off this project, but because shipping and importing from the USA to Europe is more pain than ordering from China). During the process I noticed some inconsistencies in the component descriptions in the KiCad project, the BOM file from Tindie and the actual packages on the PCB.

- Component `D1` is marked as _1N5817_, but this component as well as the linked datasheet on Tindie refer to a through-hole part. The PCB, however, clearly needs an SMD part, so shouldn't this component more likely be a _1N5819WS_?
- Component `U1` is marked as _CP2102N-Axx-xQFN28_, which has a QFN28 package with 7 pins per edge. However, the PCB has 8 pins per edge, so I would actually need a _CP2102N-Axx-xQFN24_ chip, no?

Also, a bit more information in regards to how to order the PCB at the manufacturer of your choice would be nice, although I do understand if you wouldn't like to do that because of the support effort this could create. However, you could still profit from referrals. A great example in my opinion is the [Valetudo Dreame Breakout PCB](https://github.com/Hypfer/valetudo-dreameadapter) for the [Valetudo](https://valetudo.cloud) project, where there is even a community of people sharing their PCBs (not a use-case for upsy-desky, of course) and consolidating orders for reduced shipping cost.

### How do you manage your Upsy Desky's updates?

None

tjhorner, 2024-04-01:
>Component `D1` is marked as _1N5817_, but this component as well as the linked datasheet on Tindie refer to a through-hole part. The PCB, however, clearly needs an SMD part, so shouldn't this component more likely be a _1N5819WS_?

Yes, that's correct. I'll update the BOM.

>Component `U1` is marked as _CP2102N-Axx-xQFN28_, which has a QFN28 package with 7 pins per edge. However, the PCB has 8 pins per edge, so I would actually need a _CP2102N-Axx-xQFN24_ chip, no?

No, the QFN-28 package is correct. You might have mis-counted the pins — there are 7 per edge on `U1`.

I'll see what I can do regarding more information on self-ordering a PCB from a manufacturer, but my time to work on this project is very limited at the moment. If you are ordering from JLCPCB, you can use the `LCSC` row in the BOM to correlate each part with the correct part from their inventory. Those are the exact parts I use in the pre-assembled units.

Crazyphil, 2024-04-02:
Thank you for the quick response.

> No, the QFN-28 package is correct. You might have mis-counted the pins — there are 7 per edge on `U1`.

That was my fault, I had it the other way round: The footprint was correct, but the MPN referenced a part with package QFN24 (_CP2102N-A02-GQFN24_). As I've seen from the commit, you've spotted that already.

> I'll see what I can do regarding more information on self-ordering a PCB from a manufacturer, but my time to work on this project is very limited at the moment.

I understand, time is a very precious resource indeed :smile: An alternative I came upon during my research could be PCBWay's [Shared Projects](https://www.pcbway.com/project/) that offer preconfigured public PCB (and assembly) orders and a small 10% commission to the project's author for every order. That way, users wouldn't have to deal with configuring the project for themselves. However, the minimum order amount of 5 PCBs still applies.
After comparing multiple services I can say that their offers are competitively priced, and they even offer shipping under the IOSS scheme for EU customers, which takes care of any taxes and import hassles. 

> If you are ordering from JLCPCB, you can use the `LCSC` row in the BOM to correlate each part with the correct part from their inventory. Those are the exact parts I use in the pre-assembled units.

In that regard, there's one more thing I've found which I forgot to mention: in in the BOM, the LCSC number is equal for both `R18` (1kΩ) and `R22` (940Ω). I remember darkly from my electronics course a few years ago that resistor values aren't necessarily that exact and therefore interchangeable in a certain range. However, if in this case it really doesn't matter whether to use two different components or just one, why not simplify it formally on the PCB instead of only in the BOM?

sibero80, 2024-04-29:
+1 on the PCBWAY Shared Projects idea. I'm located in South America, and getting things shipped from China vs Europe/USA is orders of magnitude easier and cheaper. I'd gladly pay 10% or 20% over PCB and parts costs if it goes to the designer.

commit 3d3ebc42 (2024-04-01): Fix error in BOM - closes #45