tjhorner/upsy-desky #46: Wireless ESPHhome Update fails, connection reset by peer

plaetzchen, 2024-06-04:
### Troubleshooting Guide

- [X] I have followed the steps in the troubleshooting guide and my issue is either not listed or is not solved.

### Description

When trying to update with the newest ESPHome version I get an error after upload that says "Connection reset by peer". The update might be installed (since the newest version is showing up in the device log) but HA still wants to install an update. Cleaning the build cache of ESPHome does not help.

### Logs

```plain text
Upsy-Desky:

INFO ESPHome 2024.5.4
INFO Reading configuration /config/esphome/upsy-desky-76db98.yaml...
INFO Starting log output from 10.0.0.94 using esphome API
INFO Successfully connected to upsy-desky-76db98 @ 10.0.0.94 in 0.068s
INFO Successful handshake with upsy-desky-76db98 @ 10.0.0.94 in 0.121s
[09:45:17][I][app:100]: ESPHome version 2024.5.2 compiled on May 21 2024, 11:25:44
[09:45:17][I][app:102]: Project tj_horner.upsy_desky version 2.0.0
[09:45:42][I][standing_desk_height:036]: Starting decoder detection
[09:45:42][I][standing_desk_height:094]: Decoder detection complete. Correct decoder variant: jarvis
[09:45:42][I][standing_desk_height:095]: If you want to make this change permanent, add the following to this sensor's configuration:
[09:46:18][W][component:157]: Component ota set Warning flag: unspecified
[09:46:18][E][component:164]: Component ota set Error flag: unspecified
[09:46:18][W][component:237]: Component ota took a long time for an operation (201 ms).
[09:46:18][W][component:238]: Components should block for at most 30 ms.
[09:46:23][E][component:176]: Component ota cleared Error flag

Build:

INFO ESPHome 2024.5.4
INFO Reading configuration /config/esphome/upsy-desky-76db98.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing upsy-desky-76db98 (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.3
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5
|-- Improv @ 1.2.3
RAM:   [==        ]  16.3% (used 53456 bytes from 327680 bytes)
Flash: [========= ]  94.5% (used 1733837 bytes from 1835008 bytes)
========================= [SUCCESS] Took 5.69 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 10.0.0.94
INFO Uploading /data/build/upsy-desky-76db98/.pioenvs/upsy-desky-76db98/firmware.bin (1739616 bytes)
ERROR Error binary size: Unknown error from ESP
```

### How do you manage your Upsy Desky's updates?

None

### ESPHome Config

```YAML
substitutions:
  name: upsy-desky-76db98
  friendly_name: Upsy Desky 76db98
packages:
  tj_horner.upsy_desky: github://tjhorner/upsy-desky/firmware/stock.yaml@v2.0.0
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: YesItHasThat

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

tjhorner, 2024-06-06:
If you are running into this error, I would recommend flashing via the USB-C port on the device. The module used on the board is pretty resource-constrained and the stock firmware has many components that may be unnecessary after first-use. For this reason, it likely runs out of RAM during an OTA flash (without inspecting the logs via USB during an OTA, it's hard to tell).

I recommend "trimming" the firmware by following this guide: https://upsy-desky.tjhorner.dev/docs/advanced/customization/#trimming-the-firmware

Then flash the resulting firmware via USB. From then on, OTAs should work.

wez, 2024-12-26:
turning off wifi power saving can also help with spotty ota: https://github.com/tjhorner/upsy-desky/issues/55#issuecomment-2423364079