tjhorner/upsy-desky #47: Breaking changes from ESPHome 2024.6.0

jhemak, 2024-06-20:
### Troubleshooting Guide

- [X] I have followed the steps in the troubleshooting guide and my issue is either not listed or is not solved.

### Description

I am not able to do OTA updates of UpsyDesky after updating to ESPHome 2024.6.0.  It appears there are multiple breaking changes that may require updates to your firmware.

https://esphome.io/changelog/2024.6.0.html

### How do you manage your Upsy Desky's updates?

OTA Updates

tjhorner, 2024-06-20:
Could you explain how you are trying to update your Upsy Desky? It sounds like you're using ESPHome Dashboard, not the OTA updates provided in the GitHub releases.

And could you share exactly what error you are receiving?

I did see the ESPHome 2024.6 changelog earlier today and will update the configs for compatibility.

jhemak, 2024-06-20:
Yes, sorry.  I used the ESPHome Dashboard and received the error below:

```
INFO ESPHome 2024.6.0
INFO Reading configuration /config/esphome/upsy-desky.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing upsy-desky (board: esp32dev; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40407.0 (4.4.7) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.4
Linking .pioenvs/upsy-desky/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/upsy-desky/src/main.o:(.literal._Z5setupv+0x3c): undefined reference to `vtable for esphome::ESPHomeOTAComponent'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/upsy-desky/src/main.o:(.literal._Z5setupv+0x4c): undefined reference to `vtable for esphome::safe_mode::SafeModeComponent'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/upsy-desky/src/main.o:(.literal._Z5setupv+0x420): undefined reference to `esphome::ESPHomeOTAComponent::set_port(unsigned short)'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/upsy-desky/src/main.o:(.literal._Z5setupv+0x424): undefined reference to `esphome::safe_mode::SafeModeComponent::should_enter_safe_mode(unsigned char, unsigned int, unsigned int)'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/upsy-desky/src/main.o: in function `esphome::Component::set_component_source(char const*)':
/data/build/upsy-desky/src/esphome/core/component.h:149: undefined reference to `esphome::ESPHomeOTAComponent::set_port(unsigned short)'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/upsy-desky/src/main.o: in function `setup()':
/data/build/upsy-desky/src/main.cpp:231: undefined reference to `esphome::safe_mode::SafeModeComponent::should_enter_safe_mode(unsigned char, unsigned int, unsigned int)'
collect2: error: ld returned 1 exit status
*** [.pioenvs/upsy-desky/firmware.elf] Error 1
========================= [FAILED] Took 49.57 seconds =========================

```
I did already change my config to include `platform: esphome` under `ota`.

Thanks, and please let me know if anything else is needed.

tjhorner, 2024-06-20:
Thanks! I will do some testing on my end and come up with a fix soon. I'll keep you updated

z00nx, 2024-06-20:
I've made this pull request to fix it https://github.com/tjhorner/upsy-desky/pull/48

tjhorner, 2024-06-20:
This is fixed in v4.0.0. Thanks @z00nx for the PR 👍

closed 2024-06-20 by tjhorner

z00nx, 2024-06-21:
> Thanks! OTA Is working now ... but does the bold below indicate I'm not getting the latest firmware? I thought it was now 4.0.0.
> 
> [18:04:12][I][app:102]: Project tj_horner.upsy_desky version **3.0.0**

I believe in the firmware that is made available for upsy desky, the ESPHome configuration is pinned to a specific release.
If you open the ESPHome configuration for your device you should see a line similar to the following
```
packages:
  tj_horner.upsy_desky: github://tjhorner/upsy-desky/firmware/stock.yaml@v3.0.0
```
You will have to update the string after `@` to the release you want to use e.g. `v4.0.0` or `master` if you just want to always use the latest development version

jhemak, 2024-06-21:
Thanks, but I had already made that change:

```
packages:
  tj_horner.upsy_desky: github://tjhorner/upsy-desky/firmware/base.yaml@master
```

tjhorner, 2024-06-21:
Yes, very strange that you're seeing the old version number. It was definitely updated to match 🤔

Maybe ESPHome is using an older version of the base config but the newer version of the stock config. That's the only explanation I can think of. What happens when you clean the build files then reflash it?

jhemak, 2024-06-21:
That did the trick!  Now I see `Project tj_horner.upsy_desky version 4.0.0`

But, FYI .. this did just flash across the screen as the only warning during the build/deploy ... not sure it matters, but wanted to let you know:

```
Compiling .pioenvs/upsy-desky/src/esphome/components/standing_desk_height/omnidesk_decoder.o
src/esphome/components/standing_desk_height/omnidesk_decoder.cpp: In member function 'virtual bool esphome::standing_desk_height::OmnideskDecoder::put(uint8_t)':
src/esphome/components/standing_desk_height/omnidesk_decoder.cpp:9:5: warning: this statement may fall through [-Wimplicit-fallthrough=]
     if (b == 0x02 || b == 0x03 || b == 0x04) {
     ^~
src/esphome/components/standing_desk_height/omnidesk_decoder.cpp:14:3: note: here
   default:
   ^~~~~~~
```