tjhorner/upsy-desky #50: Link to case files on BYOE page of the Upsy Desky site contains an extra trailing slash

nevvob, 2024-07-01:
I printed my own enclosure but notice that the link to the file on the page https://upsy-desky.tjhorner.dev/docs/byoe/ has an extra trailing slash meaning that the file does not download:

[Download STL Files](https://upsy-desky.tjhorner.dev/assets/files/enclosure-b738eb46a745e9499b85d6a1623c30b5.zip/) should be [Download STL Files](https://upsy-desky.tjhorner.dev/assets/files/enclosure-b738eb46a745e9499b85d6a1623c30b5.zip)

tjhorner, 2024-07-01:
Thanks for flagging this! It's an ongoing issue with the docs platform I use combined with the fact that it's hosted on GitHub Pages.

I'll try to find a workaround in the meantime.

nevvob, 2024-07-01:
My bad - sorry!

Thanks for creating an awesome device that worked straight out of the box.  I am genuinely impressed with the tolerances you've achieved on the printed case.

tjhorner, 2024-07-01:
Thank you for the kind words! I'm glad everything worked straight away for you :)

I just deployed a fix for this issue, seems like file download links work properly now. Thanks again for reporting it.

closed 2024-07-01 by tjhorner