tjhorner/upsy-desky #52: Warning in log when using web interface

35gh, 2024-07-03:
### Troubleshooting Guide

- [X] I have followed the steps in the troubleshooting guide and my issue is either not listed or is not solved.

### Description

The following warning are printed in logs when clicking on any button of the web interface.

### Logs

```plain text
15:54:22	[W]	[web_server_idf:070][1;31m[httpd][0;33m	

Only application/x-www-form-urlencoded supported for POST request
```

### Which version of the Upsy Desky firmware config are you running?

4.0.0

### How do you manage your Upsy Desky's updates?

ESPHome YAML Config

### If you're using an ESPHome YAML config, which version of ESPHome are you running?

2024.6.6

### ESPHome Config

```YAML
substitutions:
  name: upsy-desky-xxxxxx
  friendly_name: Jarvis Fully 2

packages:
  tj_horner.upsy_desky: github://tjhorner/upsy-desky/firmware/base.yaml@v4.0.0

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}

api:
  encryption:
    key: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
  reboot_timeout: 0s
    
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

logger:
  level: INFO

improv_serial:

web_server:
  port: 80
  version: 2
  js_url: "https://upsy-desky.tjhorner.dev/esphome-webserver/www.js"

ota:
  password: ""
  platform: esphome

external_components:
  - source: github://tjhorner/esphome-custom-components
    components: [ project_version ]

text_sensor:
  - platform: project_version
    name: "Upsy Desky Firmware Version"
    internal: true

dashboard_import:
  package_import_url: github://tjhorner/upsy-desky/firmware/stock.yaml@v4.0.0
```

tjhorner, 2024-07-03:
Can you provide any browser-side logs that appear as well? Thank you!

35gh, 2024-07-05:
Tested with Firefox 127.0.2. Let me know if you want to capture anything specific.

I see the following error in console when loading the web interface:

```
Uncaught (in promise) Error: invalid template strings array
    kt https://upsy-desky.tjhorner.dev/esphome-webserver/www.js:3
    Mt https://upsy-desky.tjhorner.dev/esphome-webserver/www.js:3
    H https://upsy-desky.tjhorner.dev/esphome-webserver/www.js:3
    _$AC https://upsy-desky.tjhorner.dev/esphome-webserver/www.js:3
    g https://upsy-desky.tjhorner.dev/esphome-webserver/www.js:3
    _$AI https://upsy-desky.tjhorner.dev/esphome-webserver/www.js:3
    It https://upsy-desky.tjhorner.dev/esphome-webserver/www.js:3
    update https://upsy-desky.tjhorner.dev/esphome-webserver/www.js:3
    performUpdate https://upsy-desky.tjhorner.dev/esphome-webserver/www.js:1
    scheduleUpdate https://upsy-desky.tjhorner.dev/esphome-webserver/www.js:1
    _$Ej https://upsy-desky.tjhorner.dev/esphome-webserver/www.js:1
    requestUpdate https://upsy-desky.tjhorner.dev/esphome-webserver/www.js:1
    _$Eu https://upsy-desky.tjhorner.dev/esphome-webserver/www.js:1
    A https://upsy-desky.tjhorner.dev/esphome-webserver/www.js:1
    f https://upsy-desky.tjhorner.dev/esphome-webserver/www.js:3
    yt https://upsy-desky.tjhorner.dev/esphome-webserver/www.js:3
    u https://upsy-desky.tjhorner.dev/esphome-webserver/www.js:3
    g https://upsy-desky.tjhorner.dev/esphome-webserver/www.js:3
    _$AI https://upsy-desky.tjhorner.dev/esphome-webserver/www.js:3
    It https://upsy-desky.tjhorner.dev/esphome-webserver/www.js:3
    update https://upsy-desky.tjhorner.dev/esphome-webserver/www.js:3
    performUpdate https://upsy-desky.tjhorner.dev/esphome-webserver/www.js:1
    scheduleUpdate https://upsy-desky.tjhorner.dev/esphome-webserver/www.js:1
    _$Ej https://upsy-desky.tjhorner.dev/esphome-webserver/www.js:1
    requestUpdate https://upsy-desky.tjhorner.dev/esphome-webserver/www.js:1
    _$Eu https://upsy-desky.tjhorner.dev/esphome-webserver/www.js:1
    A https://upsy-desky.tjhorner.dev/esphome-webserver/www.js:1
    f https://upsy-desky.tjhorner.dev/esphome-webserver/www.js:3
    N https://upsy-desky.tjhorner.dev/esphome-webserver/www.js:3
    R https://upsy-desky.tjhorner.dev/esphome-webserver/www.js:3
    R https://upsy-desky.tjhorner.dev/esphome-webserver/www.js:3
    F https://upsy-desky.tjhorner.dev/esphome-webserver/www.js:3
    <anonymous> https://upsy-desky.tjhorner.dev/esphome-webserver/www.js:3
www.js:3:335
```

Here's a request generating the warning (turn off status led):

```
{
  "log": {
    "version": "1.2",
    "creator": {
      "name": "Firefox",
      "version": "127.0.2"
    },
    "browser": {
      "name": "Firefox",
      "version": "127.0.2"
    },
    "pages": [
      {
        "id": "page_1",
        "pageTimings": {
          "onContentLoad": -1,
          "onLoad": -1
        },
        "startedDateTime": "2024-07-05T09:00:29.178+02:00",
        "title": "http://upsy-desky-xxxxxx.local/"
      }
    ],
    "entries": [
      {
        "startedDateTime": "2024-07-05T09:00:29.178+02:00",
[... 91 lines cut ...]
          },
          "redirectURL": "",
          "headersSize": 116,
          "bodySize": 580
        },
        "cache": {},
        "timings": {
          "blocked": 0,
          "dns": 0,
          "connect": 0,
          "ssl": 0,
          "send": 0,
          "wait": 262,
          "receive": 1329
        },
        "time": 1591,
        "_securityState": "insecure",
        "serverIPAddress": "192.168.1.12",
        "connection": "80",
        "pageref": "page_1"
      }
    ]
  }
}
```