tjhorner/upsy-desky #56: FRM053-3 Reporting incorrect Height

jrsecor, 2024-11-01:
### Troubleshooting Guide

- [X] I have followed the steps in the troubleshooting guide and my issue is either not listed or is not solved.

### Description

I flashed this to two separate devices. One is on a Varidesk for work and is working great. Today I tried to set another up with my Uplift for home. I have the FRM053-3 so it is listed as compatible. On first boot, it correctly identified the height as 47 inches (which is where I had it set on boot) but it started randomly going through heights in HA - I watched it jump around. It finally settled on 12.8 inches for some reason which is incorrect. 
[image]

I've hit re-detect decoder a few times to try and help it (since it stopped reporting height temporarily). IT definitely thinks it is at 12.5 as when I tried to set it to 20, it went up (I had to put weight on the desk to kick in protection and prevent some mounts from hitting the ceiling)

I'm not sure what exactly to do at this point. I can't adjust any of the yaml as ESPHome does not work in the way that the documentation states within HA. 

### Which version of the Upsy Desky firmware config are you running?

4

### How do you manage your Upsy Desky's updates?

OTA Updates

jrsecor, 2024-11-01:
It seems to settle on omnidesk on most detections, every now and then thinking it is a uplift...Problem is, I don't know how to make any detection permanent as I am fairly certain that a power loss is going to break things. 

Some logs -

`<html><body>

12:12:04 | [W] | [web_server_idf:070][1;31m[httpd][0;33m | Only application/x-www-form-urlencoded supported for POST request
-- | -- | -- | --
12:12:04 | [I] | [standing_desk_height:036] | Starting decoder detection
12:12:05 | [I] | [standing_desk_height:094] | Decoder detection complete. Correct decoder variant: omnidesk
12:12:05 | [I] | [standing_desk_height:095] | If you want to make this change permanent, add the following to this sensor's configuration:
12:12:05 | [I] | [standing_desk_height:096] | variant: omnidesk
12:12:14 | [W] | [web_server_idf:070][1;31m[httpd][0;33m | Only application/x-www-form-urlencoded supported for POST request
12:12:14 | [I] | [standing_desk_height:036] | Starting decoder detection
12:12:16 | [W] | [standing_desk_height:044] | No valid decoder found. Please make sure your desk is reporting the height and you can see it on the keypad
12:12:18 | [W] | [web_server_idf:070][1;31m[httpd][0;33m | Only application/x-www-form-urlencoded supported for POST request
12:12:18 | [I] | [standing_desk_height:036] | Starting decoder detection
12:12:19 | [I] | [safe_mode:041] | Boot seems successful; resetting boot loop counter
12:12:21 | [W] | [standing_desk_height:044] | No valid decoder found. Please make sure your desk is reporting the height and you can see it on the keypad
12:12:24 | [W] | [web_server_idf:070][1;31m[httpd][0;33m | Only application/x-www-form-urlencoded supported for POST request
12:12:24 | [I] | [standing_desk_height:036] | Starting decoder detection
12:12:24 | [I] | [standing_desk_height:094] | Decoder detection complete. Correct decoder variant: uplift
12:12:24 | [I] | [standing_desk_height:095] | If you want to make this change permanent, add the following to this sensor's configuration:
12:12:24 | [I] | [standing_desk_height:096] | variant: uplift

</body>
</html>`

jrsecor, 2024-11-01:
[logs_jacob-desk_run.txt](https://github.com/user-attachments/files/17604081/logs_jacob-desk_run.txt)

tjhorner, 2024-11-01:
It looks like you will indeed need to make the uplift decoder permanent in the YAML.

> I can't adjust any of the yaml as ESPHome does not work in the way that the documentation states within HA.

What do you mean by this? I would definitely like to fix the documentation if anything has changed. Make sure you are going to the ESPHome Dashboard addon, not the ESPHome HA integration, to adopt and edit the config.

jrsecor, 2024-11-01:
> It looks like you will indeed need to make the uplift decoder permanent in the YAML.
>
>
> What do you mean by this? I would definitely like to fix the documentation if anything has changed. Make sure [...]

I didn't have the add-on, just the integration - i fixed that and that is where that log file came from in my last reply. I set it to uplift but still am getting the jumping/incorrect data reporting