tjhorner/upsy-desky #57: JCB36NE2A1-230 - Keypad Memories do not work

PULterHO, 2025-02-11:
### Troubleshooting Guide

- [x] I have followed the steps in the troubleshooting guide and my issue is either not listed or is not solved.

### Description

Hi,
Thanks for your work on this.
Wanting to raise a bug on the above desk controller for an Ikea Mittzon.
UpsyDesky unit appears to largely work as it says on the tin, and all HA/ESPHome features work as expected.
Only issue is, the two preset buttons that came with the desk don't appear to do anything when connected to the UpsyDesky unit, and the features on the unit don't seem to modify/change anything for the keypads.

I have added   `standing_desk_variant: jarvis`, but that didn't change anything.

Thanks

### Logs

```plain text

```

### Which version of the Upsy Desky firmware config are you running?

4.0.2

### How do you manage your Upsy Desky's updates?

None

### ESPHome Config

```YAML

```

tjhorner, 2025-02-12:
Hi, thanks for raising this. Can you send a photo of your desk's keypad? Most I have seen have four presets, does yours only have two?

PULterHO, 2025-02-12:
Thanks for replying - please see attached.
The URL for the [vendor provided manual is here also](https://www.ikea.com/au/en/manuals/mittzon-underframe-sit-stand-for-desk-electric-white__AA-2415003-3-100.pdf)

[image]
[image]

DDRBoxman, 2025-08-24:
I collected some information of the keypad working with the debug Upsy Desky firmware.

Height status
`F2:F2:01:03:02:98:07:A5:7E`

Go to memory 1:
`F1:F1:05:00:05:7E:F1:F1:05:00:05:7E:F1:F1:05:00:05:7E`

Go to memory 2:
` F1:F1:06:00:06:7E:F1:F1:06:00:06:7E:F1:F1:06:00:06:7E`

After hooking up the Upsy desky the memory from the keypad stopped working.
Interestingly the memory presets that Upsy Desky uses in home assistant do work

My keypad is model: JCHT35K68C-W
The control box is: JCB36NE2A4-110

DDRBoxman, 2025-08-25:
I did a bit more tinkering and it looks like if I disconnect the TX pin (16 on the esp, 4 on the ethernet connection) everything works correctly.

I setup some test firmware to make sure nothing was going out over the TX pin and having that pin connected at all even without sending commands made presets on the remote stop working, seems like a possible hardware incompatibility.