tjhorner/upsy-desky #60: ESPHome 2025.8.0 install error

jhoff80, 2025-08-20:
### Troubleshooting Guide

- [x] I have followed the steps in the troubleshooting guide and my issue is either not listed or is not solved.

### Description

Doesn't compile with 2025.8.0.  See logs below.

Github repo is set to master, so I'm using the latest config.

### Logs

```plain text
INFO ESPHome 2025.8.0
INFO Reading configuration /config/esphome/upsy-desky-698064.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing upsy-desky-698064 (board: esp32dev; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32/releases/download/54.03.21-2/platform-espressif32.zip)
--------------------------------------------------------------------------------
INFO Package configuration completed successfully
INFO Package configuration completed successfully
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50402.0 (5.4.2) 
 - tool-cmake @ 3.30.2 
 - tool-esp-rom-elfs @ 2024.10.11 
 - tool-esptoolpy @ 5.0.2 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.13.1 
 - tool-scons @ 4.40801.0 (4.8.1) 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp-elf @ 14.2.0+20241119
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.10
|-- ArduinoJson @ 7.4.2
|-- Improv @ 1.2.4
Compiling .pioenvs/upsy-desky-698064/src/esphome/components/project_version/project_version_text_sensor.cpp.o
Compiling .pioenvs/upsy-desky-698064/src/esphome/components/text_sensor/text_sensor.cpp.o
Compiling .pioenvs/upsy-desky-698064/src/esphome/components/uart/uart.cpp.o
Compiling .pioenvs/upsy-desky-698064/src/esphome/components/uart/uart_component.cpp.o
Compiling .pioenvs/upsy-desky-698064/src/esphome/components/uart/uart_component_esp_idf.cpp.o
Compiling .pioenvs/upsy-desky-698064/src/esphome/components/uart/uart_debugger.cpp.o
Compiling .pioenvs/upsy-desky-698064/src/esphome/components/web_server/list_entities.cpp.o
Compiling .pioenvs/upsy-desky-698064/src/esphome/components/web_server/ota/ota_web_server.cpp.o
Compiling .pioenvs/upsy-desky-698064/src/esphome/components/web_server/web_server.cpp.o
Compiling .pioenvs/upsy-desky-698064/src/esphome/components/web_server_base/web_server_base.cpp.o
In file included from src/esphome/components/project_version/project_version_text_sensor.cpp:1:
src/esphome/components/project_version/project_version_text_sensor.h:14:15: error: 'std::string esphome::project_version::ProjectVersionTextSensor::unique_id()' marked 'override', but does not override
   14 |   std::string unique_id() override;
      |               ^~~~~~~~~
*** [.pioenvs/upsy-desky-698064/src/esphome/components/project_version/project_version_text_sensor.cpp.o] Error 1
```

### How do you manage your Upsy Desky's updates?

ESPHome YAML Config

### If you're using an ESPHome YAML config, which version of ESPHome are you running?

2025.8.0

### ESPHome Config

```YAML
substitutions:
  name: upsy-desky-698064
  friendly_name: Office Desk

packages:
  tj_horner.upsy_desky: github://tjhorner/upsy-desky/firmware/stock.yaml@master

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}

api:
  encryption:
    key: redacted

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  platform: esphome
  password: !secret ota_password
```

tjhorner, 2025-08-21:
Fixed here: https://github.com/tjhorner/esphome-custom-components/commit/7a7a1704113e6c63d3ab78beae12ce97e14a48c1

Please try re-compiling the firmware. If it fails, clear the build files and try again; this will force ESPHome to download the latest version of the affected external component.

VACIndustries, 2025-08-21:
Hmmm...even after cleaning it seems to be having the same issue:
```
In file included from src/esphome/components/project_version/project_version_text_sensor.cpp:1:
src/esphome/components/project_version/project_version_text_sensor.h:14:15: error: 'std::string esphome::project_version::ProjectVersionTextSensor::unique_id()' marked 'override', but does not override
   14 |   std::string unique_id() override;
      |               ^~~~~~~~~
*** [.pioenvs/upsy-desky-e16450/src/esphome/components/project_version/project_version_text_sensor.cpp.o] Error 1
========================= [FAILED] Took 21.63 seconds =========================
```

Config:
```
substitutions:
  name: upsy-desky-e16450
  friendly_name: Upsy Desky
  standing_desk_variant: uplift
packages:
  tj_horner.upsy_desky: github://tjhorner/upsy-desky/firmware/stock.yaml@master
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:

wifi:
  fast_connect: true
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  platform: esphome
  password: !secret ota_password

button:
- platform: restart
  name: "Reboot"
```

mrtnkhl, 2025-08-21:
Cleared build files, tried to install, getting the same error:

```
Compiling .pioenvs/upsy-desky-a29788/src/esphome/components/safe_mode/safe_mode.cpp.o
Compiling .pioenvs/upsy-desky-a29788/src/esphome/components/select/select.cpp.o
In file included from src/esphome/components/project_version/project_version_text_sensor.cpp:1:
src/esphome/components/project_version/project_version_text_sensor.h:14:15: error: 'std::string esphome::project_version::ProjectVersionTextSensor::unique_id()' marked 'override', but does not override
   14 |   std::string unique_id() override;
      |               ^~~~~~~~~
*** [.pioenvs/upsy-desky-a29788/src/esphome/components/project_version/project_version_text_sensor.cpp.o] Error 1
========================= [FAILED] Took 243.94 seconds =========================
```

mrtnkhl, 2025-08-21:
Will keep you posted. Probably best to wait until tomorrow to see if it updates for most people automatically without manual intervention. That way one can be sure that the problem will "fix itself" thanks to your update!

jhoff80, 2025-08-22:
Worked for me, thanks.

mrtnkhl, 2025-08-22:
Compiled again, works with no issues now.